Atmosphere-NX/Atmosphere #1765: Atmosphere enters a crash loop when there are too many erpt reports

KevinX8, 2022-01-19:
## Bug Report

### What's the issue you encountered?

My atmosphere\erpt_reports folder contained 16383 reports just from normal usage over a few years and my switch would constantly crash on boot with error 2002-0039.</br>
I made no changes to atmosphere.</br>

### How can the issue be reproduced?

Use the switch normally for about 2 years or just fill the erpt_reports folder with as many reports as you can until you get into the crash loop.</br>

### Crash Report

I deleted the crash reports in order to be able to boot my system again.</br>

### System Firmware Version

13.2.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe
- Official release or unofficial build: official 1.2.5
- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: none
- EmuMMC or SysNAND: EmuMMC file-based

### Additional context?

- Additional info about your environment: SD card is fat32

SciresM, 2022-01-19:
Yeah this isn't really an atmosphere bug, but delete your reports.

SciresM, 2022-01-19:
(this is a FAT filesystem limitation)

closed 2022-01-19 by SciresM

KevinX8, 2022-01-19:
seeing as exFAT is not recommended to be used for homebrew would it not make more sense for atmosphere to automatically delete reports when there are too many of them instead of just crashing?

SciresM, 2022-01-19:
This has nothing to do with exFAT, and "delete" if the folder is in that state, accessing it gives an error, we can't delete from that state.

SciresM, 2022-01-19:
Some erpt report cleanup can be considered, for sure, but in the meantime "just delete your reports".

KevinX8, 2022-01-19:
yes, I have deleted them, which means it is possible to delete the reports, from some brief research online the max files fat32 can have in a folder is around 65k, which 16k is nowhere near that, and also atmos could wipe the reports let's say 1k reports before the crash loop begins no so that the access problem is not a thing?

SciresM, 2022-01-19:
"accessing gives it an error" -> using the switch's FS driver :P

KevinX8, 2022-01-19:
> "accessing gives it an error" -> using the switch's FS driver :P

I understand the switch could be incapable of deleting it when the problem already began yes, which is why I suggested a preventive measure and deleting the logs when you're around 10% away from the bug occurring

SciresM, 2022-01-19:
> Some erpt report cleanup can be considered, for sure, but in the meantime "just delete your reports".

KevinX8, 2022-12-13:
I see the issue has now been fixed 🎉
https://github.com/Atmosphere-NX/Atmosphere/commit/04c9004e05facdf83ea0a351a215f761f51b9c43#diff-1a54fc072e431096621f53026a5aa1e8466e06c52f4b49cb286a1c0978b22648R100

SciresM, 2022-12-13:
Does the above actually resolve your error?

KevinX8, 2022-12-13:
I would guess so yes, I would have to accumulate the amount of logs needed to cause the crash to test this functionality which would take an extremely long time