Atmosphere-NX/Atmosphere #1767: Atmosphere keeps crashing with new fw version (13.2.1)

PixyGixy, 2022-01-23:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

 so I hadn't used my switch in a while so I updated my cfw which then I decided to update my atmosphere version, payload, sigpatches and all that. everything went fine but I went to play the new version of animal crossing that crashed whenever I booted it up, but after some research I realised I need a updated switch fw to play it so therefore I went on tinfoil to update my fw which went to daybreak and then updated my fw to 13.2.1. i rebooted my switch and I got past the atmosphere screen and the switch splash screen which then showed a black screen and after 5 seconds an atmosphere crash screen shows up, I've tried re-updating but it ends up in the same result

### How can the issue be reproduced?

detailed above ^

### Crash Report

[01642946053_01006f8002326000.log](https://github.com/Atmosphere-NX/Atmosphere/files/7920751/01642946053_01006f8002326000.log)

also for some reason the txt file was made doesn't match the time it crashed, and I sorted by latest

### System Firmware Version

13.2.1

### Environment?

directly from fusee with a rcmloader (then named it payload.bin)

- Do you have additional kips or sysmodules you're loading:

on the tinfoil screen of updating i made sure to disable all sysmodules

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

masagrator, 2022-01-23:
Piracy is not supported.

PixyGixy, 2022-01-23:
> Piracy is not supported.

u telling me that tinfoil pirates their system updates?

zemixe, 2022-01-23:
If you had read the template then you wouldnt have wasted time creating this issue:

> 
> [ Note: If the bug or crash you encountered is related to; ]
> 
> [ - software used to make "backups", ]
> [ - software explicitly distributed for piracy, etc ]
> [ then contributors will not provide support for your issue and your issue will be closed. ]

PixyGixy, 2022-01-23:
> If you had read the template then you wouldnt have wasted time creating this issue:

sorry that was my little bro, what he meant to say that he only used daybreak to update and not anything related to piracy or tinfoil and i want to fix the problem too

fincs, 2022-01-23:
Tinfoil is a piracy tool. Piracy is not supported.

closed 2022-01-23 by fincs