Atmosphere-NX/Atmosphere #1769: Issue Template version - splash.bin customization.

Claod44, 2022-01-28:
Helo again,
The python script you provided is able to add images to an existing .bin
It cannot remove or customize images already inside the bin. Can you provide a way to achieve this please ?
Thank you, have a nice day ^^

zemixe, 2022-01-28:
1. "Issue Template version"? Please do not troll the atmosphere issues tracker. You did not use the issues template for a second time, nor is this an issue with atmosphere.
2. The script directly applies the splash to package3, I have no idea what you're trying to do but it's not supported.

SciresM, 2022-01-28:
I previously linked you to a script that overwrites the image in package3. This is customization, you provide it an image and it makes it splash.

no splash is not supported, use hekate for that.

you are *still* not using the issue templates.

SciresM, 2022-01-28:
"Customize images in the bin" is what the script I linked does. It doesn't add new images, there is always exactly one splash shown in fusee.

closed 2022-01-28 by SciresM

Claod44, 2022-01-28:
I'm trying to overwrite the classic gamer splash screen, for testing purpose. 

"you are still not using the issue templates" I though you were talking about the "issues" in github. You're talking about the text defined as default message when we're writing here. Fine, I'll do next time. That's Why My issue title is "Issue Template version".

By the way, I wasn't trolling, w needed clarifications, that's all. 

About my question : So after using the script to add a slapsh screen inside Package3, it'll replace the atmosphere logo, the classic gamer logo, or both ?

SciresM, 2022-01-28:
@Claod44 the atmosphere logo. Replacing Nintendo logos is not something atmosphere has built in functionality for.

Mostly, this isn't the place for this kind of thing; the issue tracker is for tracking bugs in atmosphere, questions should be handled on discord/real time chat.

Claod44, 2022-01-28:
Not nintendo's logo. This one :
[image]

Claod44, 2022-01-28:
Can we customize it or replace it ? please ^^'

SciresM, 2022-01-28:
That logo does not appear anywhere in atmosphere. I am locking this under the presumption you're a troll.

If you're not, join the discord -- this isn't what the issue tracker is for.