Atmosphere-NX/Atmosphere #1771: Suggestion:  Some means of bypassing contents layerfs on startup

Nazosan, 2022-01-31:
This is probably most specifically relevant to when people have a home launcher theme override and do an update, but I presume it could potentially apply to other applications capable of having overrides on startup.  Currently if such an app changes with an update and there is an incompatible override that one forgets (it's all too easy to forget sometimes) to remove the user is stuck with a crash on startup and essentially a permanent bootloop.  The only solution at that point is to forcibly shut the system off, take the card out, and fix it in another machine.  Simple enough, but it can be troublesome if one is on the go for example.  It also can help streamline the process of dealing with such things in the first place (no need to even delete, just override and then redo the process after the update.)  I'm sure this could potentially apply to things other than just the launcher themes, but the bootloop crash with it alone seems worth considering some means of dealing with.

I'm thinking maybe there can be a really simple solution in the form of just simply not using layeredfs stuff on startup if some condition is met -- for example, holding R or L while starting.  (Not sure if this may require the system to be undocked and the joycons attached, but even if so it's pretty reasonably simple.)  The user can then soft-reset when they are done correcting whatever needs correcting without ever having to power off and remove the memory card.

SciresM, 2022-01-31:
...hold L? Support for this is built in to atmosphere. It's even configurable.

SciresM, 2022-01-31:
Oh, meh, wait. We specifically ignore this for pre-qlaunch. I have a code comment about it.

i actually think this isn't doable sanely but I'll think about it.

takoyaki99, 2022-01-31:
To me it’s similar to how the Vita’s CFW disables plugins on startup with holding L. So if possible, disabling sysmodules+layeredFS like that would be convenient and would lessen people asking for support on the issue tracker when their theme is weird IMO.

SciresM, 2022-01-31:
You could say only apply overrides after hid is launched but then what do you do about stuff like custom system fonts? There's bound to be problems if different system modules see different override states, and you *can't* change your mind after the fact if a sysmodule asks to read an overridden system data archive.

it's complicated. I'll be thinking on it, but to be clear there's actually not an easy way to do this.

Nazosan, 2022-01-31:
Just to be clear, it sounds almost like an either/or thought process there.  Some things do load before it could look for joycon input.  Those things can't be bypassed via holding L or whatever obviously.  Is stuff like the home launcher one of them?  If not, at least a partial solution could potentially be better than none.

BTW, besides joycon controls -- which are more ideal -- what about system buttons?  Eg vol+/- most likely (I assume power isn't ideal.)  I suppose maybe even the home button, though that requires a dongle to use obviously (though in a pinch other solutions have been found.)  Would those be handled by the same things?

SciresM, 2022-01-31:
@Nazosan it *is* either or. Ensuring system components see consistent contents is non-negotiable for ams-master.

It's a question, imo, of whether a sane way can be found that results in consistent observed contents of redirected shared resources by system modules.

SciresM, 2022-01-31:
(I would not merge something to master that allowed qlaunch to see non-redirected if e.g. sdb would if based around button presses during boot without indication of when button should be pressed)

closed 2022-01-31 by SciresM