Atmosphere-NX/Atmosphere #1774: Downgrading with daybreak stuck on Switch logo

Zontex, 2022-02-02:
## Bug Report

Downgrading with daybreak to version 11.0.0 on EMUMMC stuck on Switch logo and can't boot after reboot

### What's the issue you encountered?

My switch was on version 12.X when I first tried to downgrade the atmosphere on EMUMMC to 11.0.0, I downloaded the firmware online and extracted it into firmware/ directly on the SD card, then loaded daybreak which verified it and seems like installed successfully. once installation is complete, daybreak will ask if to reboot or shutdown, neither of these options will lead to the same result. once reboot/shutdown and load the EMUMMC again, it will be stuck on the switch logo and will require me to delete and restore from OFW again.

Later I thought it's 12.X issue so I upgraded OFW to 13.X (latest version) with latest atmosphere, hetake etc files ... and still, doing the same thing with 13.X EMUMMC doesn't work, once downgraded stuck on switch logo.

### How can the issue be reproduced?

1. make a clean EMUMMC on file
2. download firmware 11.0.0
3. put firmware into firmware/ directory on SD card
4. load daybreak and downgrade atmosphere EMUMMC

### Crash Report

N/A</br>
The system doesn't crash it's just frozen on switch logo.

### System Firmware Version

13.2.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release
  -  Official release version: latest
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: daybreak
- EmuMMC or SysNAND: EmuMMC file-based.

### Additional context?

I tried to look everywhere but mostly people upgrade from SX-OS to atmosphere and not the opposite, I never tried SX-OS so I would like to try to dual-boot them both.

SciresM, 2022-02-02:
Piracy is not supported.

closed 2022-02-02 by SciresM

Zontex, 2022-02-02:
> Piracy is not supported.

Where did I say anything about piracy? I tried to downgrade back from 13.X to 12.X or any other version, it's the same result.

SciresM, 2022-02-02:
> SX-OS

piracy.