Atmosphere-NX/Atmosphere #1776: A fatal error ocurred when running Atmosphere Program ID: 010000000000002b

didak01, 2022-02-08:
## Bug Report

### What's the issue you encountered?

After the console was on sleep, I turned it on and launched a game. Then threw me the following message:

A fatal error occurred when running Atmosphere 
Program ID: 010000000000002b
Error Desc: std:: abort() called (0xffe)

Report saved to /atmosphere/fatal_errors/report_000000000b38548a.bin
Press POWER to reboot.

After three reboots getting the same message, I read a thread where the solution posted was to delete the 'contents' folder inside Atmosphere folder which worked. After this, I updated Atmosphere to 1.2.6.

However, today happened the same. After Sleep mode, launched a game and it happened again. Inject payload didn't work either.

### How can the issue be reproduced?

After sleep mode, launch a game. Reboot with the power button will put the message in a loop. Inject payload does not work

### Crash Report
[report_000000000b38548a.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8027592/report_000000000b38548a.zip)

### System Firmware Version
EmuMMC firmware version 1.2.6

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
Fusee
- Official release or unofficial build: 
Official release version 1.2.6 (I updated atmosphere thinking that perhaps an update could solve the issue)
- Do you have additional kips or sysmodules you're loading:
JKSV, EdiZon and Emuiibo (but they were already installed more ago than a month before the crash and was working properly)
- Homebrew software installed:
EdiZon, Emuiibo, Amiigo, Goldleaf, JKSV, NXThemesInstalle and PayloadLauncher (but they were already installed more ago than a month before the crash and was working properly)
- EmuMMC or SysNAND:
EmuMMC partition-based

### Additional context?
I don't remember having done anything outside normal usage before the crash appeared; it started to appear suddenly.

masagrator, 2022-02-09:
Can you provide screenshot what you have in contents folder?

didak01, 2022-02-09:
I don’t have contents folder since I deleted it as per the solution i read, however before deletion I did a backup of atmosphere folder which actually has contents folder. Would it work?

masagrator, 2022-02-09:
> * Do you have additional kips or sysmodules you're loading:
>   JKSV, EdiZon and Emuiibo (but they were already installed more ago than a month before the crash and was working properly)

So you should answer this question: None.
JKSV and Edizon are not sysmodules and kip. Emuiibo is in contents. If you don't have contents, you don't have emuiibo

Jarvis4444, 2022-02-09:
Hey, happy to open up a new issue but I am also getting this error code based on the same environment. I have attached my contents and my fatal_errors to a ZIP folder. All was going well until I pressed my right joystick in Pokemon Arceus and Atmo won't load but Hakete 5.7.0 will and I have gone ahead and reinstalled atmosphere and made sure all payloads are correct

[error.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8033983/error.zip)

didak01, 2022-02-16:
Hi everyone. Is there anything that I can do? Im trapped in a loop :(

Skipanya, 2022-03-31:
I faced the same problem, Has anyone found a solution?  (the problem appeared from the previous version and continues in the current one).

lucasmnovais, 2022-04-02:
A friend of mine was able to avoid this error by following this guide:
https://gbatemp.net/threads/panic-occured-while-running-atmosphere-title-id-0100000000000002b.591345/

TL;DR: "(...) mount your SYSTEM partition. Delete the file /saves/80000000000000d1. Done! That file will be recreated when the switch boots."

But then, now his emunand got stuck on the NS logo

Skipanya, 2022-04-03:
I found for myself as a temporary solution to the problem, 3 days work is stable.  Creating a new EMUnand and installing firmware 13.1.0

arkeleom, 2023-01-18:
Same situation, after some hours in sleep mode, atmosphere crashes.
Trying to reboot enters in a loop of errors with Program ID: 010000000000002b, after some tries, it crashes again with program id:01000000000001000, after some more tries, it finally loads the main menu (time-date is reset to default)
using atmos 1.4.0, hekate 6.0.1 and fw 15.0.0

attached some of the reports
[reports.zip](https://github.com/Atmosphere-NX/Atmosphere/files/10450116/reports.zip)

derpaphobia, 2023-01-27:
Alright, I've also got troubles with 010000000000002b so here's mine:
I've had this issue for a while, so I decided to update atmosphere, fusee and hekate yesterday, everything worked great, switch actually felt better after that, however, now that the switch has been sleeping over night, the error came back:

1. it starts with the home button not working, pressing it does nothing, holding it down does however bring up the sleep menu as it should
2. you can bypass this by going to the wake-up screen (the press a button 3 times one) and go to the news instead, then exit out of that and it brings you back to the home screen where you can choose games
3. When you try to start a game, the switch crashes with error code: 2105-0602 (0x4b469). Press POWER button to restart.
4. Restarting it brings me back to hekate, but trying to launch atmosphere from there fails with the 010000000000002b error.

Now, I'm pretty new to the switch modding scene but please tell me if you need any more information, I'll attach some google sheets for the errors, the log file from the 010000000000002b error, the 80000000000000d1 save file, and any 010000000000002b file that I can find.

Crash:
https://docs.google.com/spreadsheets/d/1JyWXvrFwslZFFdW3iEO9Ex6RKzQDxj3k1bVqmIWCnPA/edit?usp=sharing

Start-up error:
https://docs.google.com/spreadsheets/d/1zXx36vb3JZaXAroi92UxUl8841dOu4X23Iv6Y_PmxUk/edit?usp=sharing

And there's a zip-file containing all the files I found sorted by the folder I found them in + the report.
[Switch error.zip](https://github.com/Atmosphere-NX/Atmosphere/files/10518582/Switch.error.zip)

WatchThemFall, 2023-02-21:
Exact same issue as above with the home button not working and then crashing with the same error. I don't really understand what happened since I didn't touch any files on my SD card for months and it was fine.

arkeleom, 2023-03-07:
@WatchThemFall did the error happen again?

WatchThemFall, 2023-03-07:
> @WatchThemFall did the error happen again?
no

Liambass, 2023-03-09:
@WatchThemFall How do you do the thing with the saves? It asks me to "configure keyset", so I import the file I generated with biskeydump and it tells me "0 keys imported", any idea what I'm doing wrong?

eodrambo, 2023-03-11:
@watchthemfall 

> Exact same issue as above with the home button not working and then crashing with the same error. I don't really understand what happened since I didn't touch any files on my SD card for months and it was fine.
> 
> Edit: Seems I fixed it by using [NxNandManager] to delete the save file 80000000000000d1 from my NAND, and replace the save files 8000000000000050 and 8000000000000052 with the ones from my original NAND backup as I read that deleting these save 2 system save files caused a black screen for some people. Also I updated from firmware 15.0 to 15.0.1 just to refresh the firmware. Not sure what fixed it as I did all 4 at once, but it seems good now. Hard to test as it takes a full day/s for the error to show up.

This worked for me after experiencing the issue mentioned. 

Had a slight issue with my nand dump from NXNandManager, needs to be a short string and not too deep in the file tree to be recognized as nx storage. best to operate at the root of the drive.

Juemo1010, 2023-03-17:
> Exact same issue as above with the home button not working and then crashing with the same error. I don't really understand what happened since I didn't touch any files on my SD card for months and it was fine.
> 
> Edit: Seems I fixed it by using [NxNandManager] to delete the save file 80000000000000d1 from my NAND, and replace the save files 8000000000000050 and 8000000000000052 with the ones from my original NAND backup as I read that deleting these 2 system save files caused a black screen for some people. Also I updated from firmware 15.0 to 15.0.1 just to refresh the firmware. Not sure what fixed it as I did all 4 at once, but it seems good now. Hard to test as it takes a full day/s for the error to show up.

> Exact same issue as above with the home button not working and then crashing with the same error. I don't really understand what happened since I didn't touch any files on my SD card for months and it was fine.
> 
> Edit: Seems I fixed it by using [NxNandManager] to delete the save file 80000000000000d1 from my NAND, and replace the save files 8000000000000050 and 8000000000000052 with the ones from my original NAND backup as I read that deleting these 2 system save files caused a black screen for some people. Also I updated from firmware 15.0 to 15.0.1 just to refresh the firmware. Not sure what fixed it as I did all 4 at once, but it seems good now. Hard to test as it takes a full day/s for the error to show up.

I want to know how to extract the save files 8000000000000050 and 800000000000005 from the back up?

arkeleom, 2023-03-25:
Could someone post a step-to-step guide of the process? (Or a link about where to find one)

uuganaag, 2023-03-31:
Same issue here, temporarily fixed and error comes back after few days of use
followed https://github.com/Adubbz guide on https://github.com/Atmosphere-NX/Atmosphere/issues/1256#issuecomment-739803986
generated and restored PRODINFO following this guide https://www.youtube.com/watch?v=gpjkXtHR8Jk
deleted 80000000000000d1 and copied over 8000000000000050 and 8000000000000052 from backup

macche80, 2023-04-20:
I just had the same issue today, in my emunand.
From where I can get the 8000000000000050 and 8000000000000052 files? My sysnand is still working, but it's several fw version below my emunand. Can I get those files from my sysnand? And if yes, how to exactly do that?

chipplepop, 2023-04-29:
Same issue as everyone else here. Tried replacing the files and deleting the other, no luck so far. At this point i may need to cut my losses and wipe it all and begin again.

hero3210, 2023-06-16:
> Same issue as everyone else here. Tried replacing the files and deleting the other, no luck so far. At this point i may need to cut my losses and wipe it all and begin again.

same issue as you. Please reply if you were able to solve it.

Suddenly while playing BOTW I had a system freeze (no buttons would work), suddenly the Switch rebooted. and I got the same error massage as everyone here.
Tried solving it using NxNandManager but whenever I try to copy any files windows wouldn't allow me due to the name being too long (tried many things, wasn't able to solve it). So I used HacDiskMount instead and that worked. I deleted 80000000000000d1 from the SYSTEM partition in my emuNAND and replaced 8000000000000050 and 8000000000000052 but I AM STILL UNABLE TO FIX IT. I am still getting the exact same error.

Please help anyone.

eodrambo, 2023-06-16:
> same issue as you. Please reply if you were able to solve it.
>
> Suddenly while playing BOTW I had a system freeze (no buttons would work), suddenly the Switch rebooted. and I got the same error [...]

even though i was able to fix the issue in the short term, certian games refused to launch, and after a while in sleep mode the system would refuse to come back. 

what finally solved the problem is re making my emuEmmc and starting fresh. my thought is that this stems form a bad reboot/ corruption on the sd card, and that can just randomly happen.

chipplepop, 2023-06-17:
> even though i was able to fix the issue in the short term, certian games refused to launch, and after a while in sleep mode the system would refuse to come back.
>
> what finally solved the problem is [...]

Same here. I had no other choice as the crash happened every single time after a certain point and there was absolutely no getting past the crash message, and I tried _everything_. I also highly suspect it's just corruption - the files that were affected in my logs always pointed to player select screen (I have never installed themes), so this time I opted out of my second player profile to help reduce whatever variables might be involved in whatever way I can.

vamidi, 2023-07-23:
The fix of removing the 80000000000000d1 file inside the SYSTEM partition and copying these files 8000000000000050 & 8000000000000052 over from my old backup to the corrupted one helped me.

Here is the [link](https://alvinr.ca/2023/05/21/fixing-a-fatal-error-occured-when-running-atmosphere-title-id-010000000000002b/) that helped me do it.

cofw2005, 2024-10-27:
Want to add my 2 cents here.
I met same issue today and I found this post. Luckily I backed up before my sysNand and emuNand, what crashed is my emuNand atmosphere. I retrieved the there files (d1, 50 and 52) out of my backup emuMMC.
In the end in my case, deleted 80000000000000d1 caused black screen as above some mentioned. So I just replaced all three save files (d1, 50 and 52). It solved my crash.

closed 2025-04-03 by SciresM

AlbertDuong, 2025-08-16:
Hi everyone, i have this issue

A fatal error occurred when running Atmosphère.
Program ID: 010041544d530000
Error Desc: std::abort() called (0xfffe)
Report saved to /atmosphere/fatal_errors/...
Press POWER to reboot.

Can you please help me how to solve it? Thank you!