Atmosphere-NX/Atmosphere #1778: Homebrew crash after 30s; HOS 4.0.1 + Atmosphère 1.2.6

kkaazzee, 2022-02-12:
## Bug Report

### What's the issue you encountered?

Launched homebrew from hbmenu causes crash after 30s of black screen. Hbmenu launched by holding R while launching gamecard. Crash handler screen displays Error Code: 2003-0004 (0x803). This may very well be a homebrew loader/menu issue but I thought it interesting to file here.

Reproduced with daybreak, nx-shell, reboot-to-payload.

No changes were made to Atmosphère itself.

### How can the issue be reproduced?

1. Install Horizon 4.0.1, in my case I restored a clean NAND backup. Burnt fuses match 4.0.1 (5 - 0 HOS: 4.0.0 - 4.1.0).
1. Boot Atmosphère via Hekate; [`hekate_ipl.bin`](https://gist.github.com/kkaazzee/58069e1121f96c7583231f2d009c63ca).
1. Hold R while launching game card.
1. Select daybreak, nx-shell, reboot-to-payload.
1. Homebrew launches to black screen for approximately 30s.
1. Crash report displayed.

Anecdotes:

* Launching daybreak, nx-shell, reboot-to-payload from _applet mode_ succeeds. Eg; no crash.
* Rebooting to payload works fine from the crash handler.

### Crash Report

[crash_reports.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8054554/crash_reports.zip)

### System Firmware Version

HOS 4.0.1

### Environment?

- Erista, unpatched.
- Re-formatted fat32 SD, clean manual [setup](https://rentry.org/EristaSysNAND). See [files listing](https://gist.github.com/kkaazzee/538cc7225409af3f82c1a323b4caa1ec).
- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate [5.7.0](https://github.com/CTCaer/hekate/releases/tag/v5.7.0)
- Official release or unofficial build: Official
  - Atmosphere/hbmenu: [1.2.6](https://github.com/Atmosphere-NX/Atmosphere/releases/tag/1.2.6)
  - NX-Shell: [3.2.0](https://github.com/joel16/NX-Shell/releases/tag/3.20)
- Do you have additional kips or sysmodules you're loading: [patches](https://github.com/ITotalJustice/patches/releases/tag/13.2.1-1.2.6)
- Homebrew software installed: daybreak, reboot-to-payload, choidujournx, nx-shell, awoo, gold leaf
- EmuMMC or SysNAND: SysNAND

### Additional context?

Thanks for reading. I don't suppose there are many reasons to be running latest Atmosphere on an older HOS. In my particular case I planned to use Daybreak to upgrade to 13.2.1. I will try:

- daybreak in applet mode
- restoring later HOS version NAND backup (10-11 ish) then retry daybreak in !applet mode
- setup atmosphere's release for 4.0.1; [0.7.1](https://github.com/Atmosphere-NX/Atmosphere/releases/tag/0.7.1)?

Thanks for reading!

kkaazzee, 2022-02-12:
Follow-up: Homebrew which failed above launches correctly after updating HOS to 13.2.1 via daybreak in applet mode.

SciresM, 2025-04-03:
This was os::ResultBusy(), no idea, but closing as it was noted that the issue doesn't occur on newer firmware.

closed 2025-04-03 by SciresM