Atmosphere-NX/Atmosphere #1782: Audio Crash Error crashes at random intervals, software or hardware?

Hijacc, 2022-02-18:
## Bug Report

[ * ]</br>
[ software used to make "backups", ]
[want to have sys-clock to monitor temps]</br>

### What's the issue you encountered?

Describe the issue in detail and what you were doing beforehand:

Launch Atmosphere then launch game title at random intervals it will crash, I'm honestly just trying to inquire about the nature of the crash I know it's pointing towards an audio error but wonder if anyone can shed light on why it's happening. software or hardware related. It's only happening in docked mode but temps appear fine and cool to touch.

### How can the issue be reproduced?

[ * ]</br>
[When launching titles in CFW at random intervals does not occur in vanilla (or at least have not been able to reproduce it) do not currently have any custom sysmodules or themes.]</br>

### Crash Report

[ https://gist.github.com/Hijacc/8a43d7289fa380d868dbcf1f11f3ec8d ]</br>

### System Firmware Version

13.2.1</br>

### Environment?

- Injected via Tegra, Fusee.bin, Hekate built EmuMMC
- Official release or unofficial build:  [ Ver. 1.2.6 Official build]
- Do you have additional kips or sysmodules you're loading: [N/A]
- Homebrew software installed: [ N/A]
- EmuMMC 
  - Partition based on Micro SD in EmuMMC

SciresM, 2022-02-18:
Post a screenshot of your atmosphere/contents subdir?

Hijacc, 2022-02-18:
[image]

I don't currently have that.

SciresM, 2022-02-18:
Huh. I've got nothing, atmosphere by default does literally nothing audio related.

Does error occur with totally stock ams config(?), are you doing anything audio related?

Gonna be honest and say I don't *think* this is likely to be ams related but yeah I've got nothing.

Hijacc, 2022-02-18:
In stock it doesn't do it, If I boot into the stock firmware and run atmosphere it does "an error has occured" to home screen. If launch it through emuMMC it crashes with this error.

That's the oddity of it.

SciresM, 2022-02-18:
Huh. Does it vary sysmmc/emummc?

Hijacc, 2022-02-18:
I'm a little fuzzy on some of the terminology I'll try my best to explain.

If I run the vanilla software off of the internal memory on switch(sysnand?), the issue so far does not occur but I can continue to test it. 

If I run EmuMMC with CFW I get the above error anywhere from 40 minutes to an hour into a game consistently (note this error only occurs during game processes), if I run the sysnand with CFW then I get "error has occured" and crash to home screen  but don't receive an error code and can just re-open the application. I have not tried to run the system vanilla off of the SD card on its own.

All games are currently located on the internal memory.

The oddity is that it always points to audio, except when it just crashes to home screen I know with alot of common sysmodules it causes more resources to be used and I assume causes the system to run out of memory and error out.

If it was indeed the ram I feel it should be happening in the stock environment aswell but mayhaps the emulated software is using more ram? I do not know.

I suspected maybe it was over heating but before I completely started from scratch I used sys-Clock to check temperatures and never had it rise above 60c with it averaging 45c.

Hijacc, 2022-02-18:
I have never ran the stock OS off the the memory card only the emulated one.

When you say related to "dsp"  that means?

SciresM, 2022-02-18:
dsp -> sound processor.

"never ran the stock OS" ...and you're unwilling to do so for testing...?

Hijacc, 2022-02-18:
No, I can.

I just wasn't aware that I could (move?) The stock os to the memory card.

SciresM, 2022-02-18:
I don't know what you mean by "move the stock os to the memorycard".

I was suggesting booting system nand (off internal memory) with atmosphere.

Hijacc, 2022-02-18:
Oh!

I did a poor job explaining, no doubt to my loose understanding.

I have done that. That's what causes the "error has occured" issue but I do not receive an error code and I don't have to hard reset.

Hijacc, 2022-02-18:
Doing so now.

closed 2025-04-03 by SciresM