Atmosphere-NX/Atmosphere #1784: Achievement Support

david2278, 2022-02-22:
## Feature Request

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ If you do not have the information needed for a section, replace its contents with "Unknown". ]</br>

### What feature are you suggesting?
#### Overview:
- Add a barebones achievements framework</br>

#### Smaller Details: 
- This would be a framework set up for other developers to come in and make achievements on a per-game basis</br>

#### Nature of Request:  
- Addition
  - Icon at the bottom of the home screen which opens the Achievements UI
  - A user interface for Achievements.
  - A hook system which watches for specific values in memory which when found would unlock an Achievement
  - Documentation for other developers to create achievements for Switch games
### What component do you feel this would best fit within?
Unknown, likely Troposphère

### Why would this feature be useful?
This would extend the enjoyment of our favorite Switch games</br>
This would build a community of developers around building achievements for Switch games</br>

SciresM, 2022-02-22:
This is deeply out of scope for atmosphere.

SciresM, 2022-02-22:
Feel free to make something like this, though!

closed 2022-02-22 by SciresM

david2278, 2022-02-22:
Should this not exist at the OS level? Even if this was made as a pull request and merged into the project?

SciresM, 2022-02-22:
@david2278 It's very, very out of scope for Atmosphere. I would not want to be responsible for maintaining something like that, personally -- the whole point of my reimplementation philosophy is to avoid having to maintain tons of patches for official binaries.

Atmosphere does give enough invasive control of the system for something like this to be made -- I see no reason for it to be integrated into Atmosphere, when such a system could instead be maintained by its creator.

david2278, 2022-02-22:
Ok fair enough. Thanks.