Atmosphere-NX/Atmosphere #1786: Panic Occurred while running Atmosphére

kronictripy420, 2022-02-25:
Bug Report

"Panic Occurred while running atmosphere

Title ID: 010000000000002B
Error: std: :abort (0xFFE)

Report saved to /atmosphere/fatal_errors/report_0d1c4788.bin

Press POWER to continue."

What's the issue you encountered?
Hello SciresM let me start by saying thanks for all of your hard work on this. ok i was using sx os 2.9.2 and OFW 9.0.1 no emunand or sysnand back up for the most part now there dead. is it ok to update OFW to 13.2.1 then put files from atmosphere and hekate, sigpatches on sd card and start emuMMC, which one is better sd partition or sd files. Well i just updated to OFW 13.2.1 put all files from Atmosphere, Hekate, and sigpatches, also found 2 sigpatches on another form that were not included in the gethub section for sigpatches. At first i couldnt get Atmosphere or Hekate, was just giving my Fatal error every time i tried to launch Atmosphere, because of the Hakate_ipl.ini, had to create it n paste in .txt and had to save it in different file extention (.ini) and now got Hekate working, now trying to load Atmosphere was getting Fatal Error, now im getting Panic Occured while Running Atmosphere and that all im getting could you help me please?

How can the issue be reproduced?
inject hekate or just fuees.bin and let it load will recreate my issue, also cant load emuMMC or sysMMC either one will not load Atmosphere 

Crash Report
N/A

System Firmware Version
13.2.1

Environment?
- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- I used tegrarcmgui to send hekate payload, then tried to launch Atmosphere, also have just sent fuees.bin and just get Panic error or Fatal error.

- Official release or unofficial build:
- Offical release 1.2.6, all files obtained from Gethub, even the 2 sigpatches
 
- Do you have additional kips or sysmodules you're loading:
- I dont believe so 

- Homebrew software installed:
- EmuMMC or SysNAND: 
- i want both but i have atm emuMMC/ its in partition based
  
Additional context?
N/A

kronictripy420, 2022-02-25:
the gist is not letting me upload any fatal error files so ill post them here if this is ok
report_0d1c4788.bin
this the most resent one 
report_07473a8a.bin

kronictripy420, 2022-02-25:
i can also teamviewer with someone willing to help me so they can see if i have the correct files on pc and look at my fatal error codes

HookedBehemoth, 2022-02-25:
You didn't actually attach the files. You can zip them and attach them here directly.

kronictripy420, 2022-02-25:
[Fatal Errors.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8141160/Fatal.Errors.zip)

kronictripy420, 2022-02-25:
really thanks

kronictripy420, 2022-02-25:
helps alot

kronictripy420, 2022-02-25:
@HookedBehemoth could you help me with my issue?

HookedBehemoth, 2022-02-25:
Are you on MacOS?
If so, locate the "fix archive bit" option in hekate.

And never plug that SD card in your mac again.

kronictripy420, 2022-02-25:
no i hate Macos its trash im on windows 10 home

kronictripy420, 2022-02-25:
i just did that anyway and this is what Hekate is saying

Traversing all SD card files
This may take saome time...

/switch

Total archive bits fixed: 30 unset, 0 set!

kronictripy420, 2022-02-25:
Dude thanks a million i have a 256gb card with my emuMMC on it and everything else did what u said and now it works thanks

closed 2022-02-25 by SciresM