Atmosphere-NX/Atmosphere #1792: A fatal error occurred when running Atmosphère. / Program ID 010000000000002b / Error Desc: std::abort() called (0xffe)

bejamartins, 2022-03-12:
## Bug Report

Can no longer boot to Atmosphère

A fatal error occurred when running Atmosphère.
Program ID 010000000000002b
Error Desc: std::abort() called (0xffe)

### What's the issue you encountered?

The console soft rebooted once on itself and started giving this error. I can no longer boot.
I tried to update Atmosphère to the latest version.

### How can the issue be reproduced?

Booting into Atmophère through Hekate.

### Crash Report

Lastest crash report (I think):
[01646108171_010000000000000c.log](https://github.com/Atmosphere-NX/Atmosphere/files/8237141/01646108171_010000000000000c.log)

### System Firmware Version

11.0.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build:  1.2.6-master
- EmuMMC partition-based.

REXREGIUM, 2022-03-24:
Hello friends, the same thing happens to me.

My case is that I was on Fw 13.2.1 with emunand and atmosphere 1.2.6 and everything worked perfectly. Today I went to update the Fw to 14.0.0. I do the steps as usual and it updates fine, but when I restart the atmosphere already updated to v.1.3.0, it gives me the same error as in the post. and there is no way to fix it.
I don't know what to do and my son rushing me to play!

Poor me!!!

Masamune3210, 2022-03-24:
Make sure you have a completely clean install of Atmosphere, no piracy materials, no sysmodules other than whats bundled. Likely either corrupted install or sysmodules malfunctioning due to the new update not being supported yet.

REXREGIUM, 2022-03-24:
> Asegúrese de tener una instalación completamente limpia de Atmosphere, sin materiales de piratería, sin módulos de sistema que no sean los incluidos. Es probable que la instalación esté dañada o que los módulos del sistema no funcionen correctamente debido a que la nueva actualización aún no es compatible.

Thanks friend for answering.
I don't have anything in the Atmosphere folder, only what comes when you download it, I don't have any application or anything, just the folders that come with atmosphere.

So I understand that I should wait for atmosphere to be fully functional for FW 14.0.0?

Thanks for the trouble, friend.

Masamune3210, 2022-03-24:
14.0.0 is already supported, so unless you are running into a edge case you shouldn't have any issues. Unfortunately you will have to wait until someone who can help more shows up

Spade393, 2022-03-24:
Make sure that if you have a contents folder to delete the folder that has the same ID as the error code.

REXREGIUM, 2022-03-24:
Friends! I'm not very prepared on the nintendo switch, what I did was following a youtube tutorial and the first time it went well, and the second time it was this, it went wrong.
I understand English regularly, I defend myself with the translator.

I have downloaded atmosphere 1.3.0 atmosphere-1.3.0-prerelease-b69fa1357+hbl-2.4.1+hbmenu-3.5.0 and copy it to the root and overwrite it, but I still get the same error.

In the root of the SD I have the following folders:
atmosphere
boot loader
config
emuMMC
Nintendo
nsp
payloads
retro-arch
roms
switch
hbmenu.nro
retroarch.jpg

And inside the atmosphere folder I have a lot of folders, they are these:
automatic_backups
config
config_templates
contents
crash_reports
erpt_reports
exefs_patches
fatal_errors
fatal_reports
flags
hbl_html
hosts
kip_patches
logs
update
hbl.nsp
package3
reboot_payload.bin
stratosphere.romfs

I don't know if it is a file that creates some incompatibility or something, I don't know, I have deleted a folder with a number and letters that was the name of an error, it was inside the CONTENTS folder. I'll try to see if it works now.

Thank you for the trouble you are taking to help me.

REXREGIUM, 2022-03-24:
> Make sure that if you have a contents folder to delete the folder that has the same ID as the error code.

IT WORKS!!!!!!

Thank you very much, the key was that friend, delete the folder that was inside CONTENTS and that's it, problem solved, and my Switch updated to FW 14.0.0.

Really thank you very much friend! My son is very happy.

closed 2022-03-24 by SciresM

bejamartins, 2022-03-27:
@Spade393 @SciresM 
Deleting the /atmosphere/contents/010000000000002B doesn't solve my problem, simply gives me a black screen with no error.

Masamune3210, 2022-03-27:
Then you have something else causing atmosphere to not function correctly, just in case since you didn't answer the first time, back up your atmosphere directory to your computer or somewhere else other than your sd card and extract a clean version of atmosphere with no modifications to your sd card just to see if it will boot

bejamartins, 2022-03-27:
> Then you have something else causing atmosphere to not function correctly, just in case since you didn't answer the first time, back up your atmosphere directory to your computer or somewhere else [...]

@Masamune3210 did that just now, clean Atmosphère and clean Hekate and it gives me back the same error. I mounted my emuNAND with NxNandManager and removed /save/80000000000000d1, did nothing as well.

REXREGIUM, 2022-03-27:
> > Entonces tiene algo más que hace que la atmósfera no funcione correctamente, por si acaso, ya que no respondió la primera vez, haga una copia de seguridad de su directorio de la atmósfera en su computadora o en otro lugar que no sea su tarjeta SD y extraiga una versión limpia de la atmósfera sin modificaciones a su tarjeta SD solo para ver si arranca
> 
> @ Masamune3210 hizo eso hace un momento, limpió Atmosphère y limpió Hekate y me devuelve el mismo error. Monté mi emuNAND con NxNandManager y eliminé /save/80000000000000d1, tampoco hice nada.

friend, try to also delete everything in the FATAL_ERRORS folder. See if that works for you.

bejamartins, 2022-03-27:
> friend, try to also delete everything in the FATAL_ERRORS folder. See if that works for you.

Didn't work, same error :( Generated a new BIN:
[report_0c6c92ff.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8358066/report_0c6c92ff.zip)

krtpowa, 2023-03-31:
> ¡Amigos! No estoy muy preparado en el Nintendo Switch, lo que hice fue seguir un tutorial de youtube y la primera vez salió bien, y la segunda vez fue esto, salió mal. Entiendo inglés regularmente, me defiendo con el traductor.
> 
> He descargado atmosphere 1.3.0 atmosphere-1.3.0-prerelease-b69fa1357+hbl-2.4.1+hbmenu-3.5.0 y lo copio en la raíz y lo sobrcribo, pero sigo recibiendo el mismo error.
> 
> En la raíz de la SD tengo las siguientes carpetas:
> 
> atmosphere boot loader config emuMMC Nintendo nsp payloads retro-arch roms switch hbmenu.nro retroarch.jpg
> 
> Y dentro de la carpeta atmosphere tengo muchas carpetas, son estas: automatic_backups config config_templates contenidos crash_reports erpt_reports exefs_patches fatal_errors fatal_reports flags hbl_html hosts kip_patches logs update hbl.nsp package3 reboot_ Carga útil.bin stratosphere.romfs
> 
> No sé si es un archivo que crea alguna incompatibilidad o algo así, no sé, he borrado una carpeta con un número y letras que era el nombre de un error, estaba dentro de la carpeta CONTENIDO. Intentaré ver si funciona ahora.
> 
> Gracias por la molestia que estás tomando para ayudarme.
> 
> **FUNCIONA!!!!!!
> 
> **Muchas gracias, la clave era ese amigo, borrar la carpeta que estaba dentro de CONTENIDOS y listo, problema resuelto, y mi Switch actualizado a FW 14.0.0.**
> 
> Realmente muchas gracias amigo! Mi hijo está muy feliz.**

On my case, erase the content of fatal_error directory and now works!

wicho108, 2023-10-08:
Hice el mismo procedimiento y funciono, borre el archivo y encendio, pero actualice el firmware, luego al reiniciar se quedo la pantalla en negro, y ahora que intento desde TegraRcm igual no me da error y se queda la pantalla en negro, ayuda!!!!!!!!!

DCaedus, 2023-12-11:
> IT WORKS!!!!!!
>
> Thank you very much, the key was that friend, delete the folder that was inside CONTENTS and that's it, problem solved, and my Switch updated to FW 14.0.0.
>
> Really thank you very [...]

This worked for me too, it was a sysmodule causing the issue, apparently.