Atmosphere-NX/Atmosphere #1795: [Feature Request] automatically delete old erpt reports

DraconicNEO, 2022-03-16:
## Feature Request
 
 
### What feature are you suggesting?
#### Overview:
 
Have Atmosphere delete erpt reports automatically after a certain amount has been reached in the folder to prevent crashing from too many erpt reports.
 
#### Smaller Details: 
- checking number of erpt reports
- Deleting old ones to prevent them from exceeding some amount i.e. 1024
 
#### Nature of Request:  
[ Remove all that do not apply to your request. ]
- Addition
- Change
 
 
### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
 
### Why would this feature be useful?
 
This feature addresses an issue where the erpt reports accumulate over time eventually exceeding the Switch's file limit and causing the error applet to crash. If this feature were implemented this crash could be avoided since old Erpt reports would be deleted and would never accumulate to the 16383 file limit.

SciresM, 2022-03-17:
Unless a setting is switched away from default, erpt will now clean the report folder on boot if there are >= 1000 reports (or if reading the number of reports fails).

closed 2022-03-17 by SciresM

ImanCol, 2022-03-22:
Or something better, disable creating logs, if the user is not reporting bugs. In addition, it will increase the life cycle of the MicroSD.

Masamune3210, 2022-03-22:
Logs are on for a reason, issues don't always manifest before they become an imperative

SciresM, 2022-03-22:
> Or something better, disable creating logs, if the user is not reporting bugs. In addition, it will increase the life cycle of the MicroSD.

I am not going to disable log creation.

Masamune3210, 2022-03-22:
This will prevent the folder from growing past a certain point, and just because you arent getting full crashes doesn't mean errors arent happening. erpt isn't going to just spit out a log saying All is OK, if its making logs its for a reason

ImanCol, 2022-03-22:
Well, hopefully that option to delete records when it exceeds a certain amount is added.

Masamune3210, 2022-03-22:
The commit is already done, unless something changes its happening
[image]

commit 9a38be20 (2022-03-17): erpt: add automatic report cleanup mechanism on boot (closes #1795)