Atmosphere-NX/Atmosphere #1797: 14.0.0 Support

SciresM, 2022-03-22:
Master issue/check list of stuff to do for 14.0.0 changes.

- [ ] Reverse engineer new kernel changes 
- [ ] Determine what system modules need reverse engineering changes
- [ ] ([Placeholder for reverse engineering sysmodules which need reverse engineering, to be edited later...])
- [ ] Update fusee/exosphere with new keydata/values
- [ ] Implement new changes into mesosphere
- [ ] ([Placeholder for implementing changes in sysmodules discovered to be necessary via reverse engineering, to be edited later...])
- [ ] Update emummc
- [ ] Make sure libnx is updated for anything relevant.

Post-release:
- [ ] Reverse engineer potential new boot sysmodule changes.
- [ ] Miscellaneous post-release reverse engineering
- [ ] Miscellaneous post-release implementation work/updating

Linklyons, 2022-03-22:
Anything special in this one? Or does this look like a quick n easy update?

yuowen, 2022-03-22:
Request an update as soon as possible Automatically updated

SciresM, 2022-03-23:
Updated issue with summary of tasks. It's mainly just emummc left, though there's also a little misc dram id bullshittery, and I'm still technically finishing up meso. All dev work for meso is done, but I need to adjust some codegen tweaks that were put in place to make -Os better now that we're -O3 again.

SciresM, 2022-05-06:
Note: Finished going through boot sysmodule today.

Only notable code changes were in pwm, which removed the Set/Get Duty accessors in favor of the newer Scale accessors.

Note that I did not delete the older accessors, even though Nintendo did; not sure what to do about those.

In any case, with loader/boot changes reversed (and all hardware parameter scripts run with no changes detected), I'm satisfied we have what I would call full 14.0.0 support.

commit 4db48508 (2022-05-06): strat: update for code changes found in boot (closes #1797)
closed 2022-05-06 by SciresM