Atmosphere-NX/Atmosphere #1798: After Updating to Atmosphere 1.2.6 Fatal Error occured and the exFat Partition gets destroyed

ferris1000, 2022-03-22:
## Bug Report

A fatal error occurred when running Atmosphère.
Program ID: 010041544d530000
Error Desc: std::abort() called (0xffe)

exFat Partition needs to be formated after this error occurred.

### What's the issue you encountered?

After Updating to Atmosphere 1.2.6 the Error occured and the exFat Partition gets destroyed.

### How can the issue be reproduced?

Format an SD Card to exFat and try to Boot Atmosphere,

### Crash Report

Due to corrupted Partition it's impossible to get a log File, 

Chkdsk /F fixes the SD Card, but no Log.

### System Firmware Version

13.2.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Trinket based Modchip
- Official release or unofficial build: amsPlus 1.2.6 (173d5c2d3)
- Do you have additional kips or sysmodules you're loading: No
- EmuMMC or SysNAND: SysNAND

### Additional context?

The Partition Corruption already happened to 2 Consoles from 2 Friends I helped to Update their Switch.
Only exFat is effected, reformatting to exFat repeats the same Issue. 

[image]

SciresM, 2022-03-22:
Yep. Don't use exFAT, it's a nintendo driver issue.

SciresM, 2022-03-22:
Use FAT32.

closed 2022-03-22 by SciresM

ferris1000, 2022-03-23:
I'm aware that exFat sucks on Nintendo Switch that's why I'm using Fat32 myself, but some other People who are not Technical are buying a Second microSD Card and doesn't Care as much and that's my we Found the Bug the Hard Way.

SciresM, 2022-03-23:
I mean, that's irrelevant. There's nothing we can do about it except tell you to use fat32.

ferris1000, 2022-03-23:
I already did, Atmosphere worked before with exFat, only since the latest Version this happened and I thought that might be new. Now I know better, thanks.