Atmosphere-NX/Atmosphere #1800: Auto Update for atmosphere

thejocraftkit, 2022-03-23:
So I have been using atmosphere 4 years now and every time there was a system update I would need to download the latest version of atmosphere and install it.

Now the time has come for me to say it's annoying because I am always waiting till the newest version of atmosphere before I update the switch.
That's why I am requesting a new feature. It should be called "Update to the newest Version of Atmosphere" 

This feature should be able to download the newest version of atmosphere and install it after it did that it should turn on  you'll see the message Fuseé couldn't be used or whatsoever now if there is a message before you update the version please insert RCM Module it should boot to the RCM mode from the PC you can easily inject the Payload and the switch 

Feature Name: "Update Atmosphere"
Popup Message: "Please insert RCM Module" with a OK button.
Features: Download and install Newest version of Atmosphere; Boot to RCM.
Manual Downloads: newest version of Fuseé

SciresM, 2022-03-23:
No.

SciresM, 2022-03-23:
We don't support on-device updates, updates should be performed on pc.

closed 2022-03-23 by SciresM

Masamune3210, 2022-03-23:
Too much variability to make it safe anyway

thejocraftkit, 2022-03-23:
Why

thejocraftkit, 2022-03-23:
Can you make it a Programm?

SciresM, 2022-03-23:
No.

takoyaki99, 2022-03-23:
Just use AIO Updater for that feature, it’s not something the Atmos devs have said they want to do, they’ve already said that in-device updating isn’t the best…

shadowninja108, 2022-03-23:
Every modern OS includes a ZIP extractor, there’s your installer.