Atmosphere-NX/Atmosphere #1802: A fatal error occurred when running Atmosphère. / Program ID 010000000000bd00 / Error Desc: std::abort() called (0xffe)

SasorizaTsugare, 2022-03-24:
## Bug Report
### What's the issue you encountered?
Unable to Boot 1.3.0 (pre-release), I just updated it to test on my 14.0 switch firmware. I copied the contents of the release .zip to the root of my sd card and overwritten the conflicting files. I also replaced all of my fusee.bin on both my computer and in my bootloader folder with the one available beside the new atmosphere. Issue occurs with both hekate launch (5.7.2) and tegrarcmgui launch (2.6).

### How can the issue be reproduced?

replace existing atmosphere and fusee.bin files with the new pre-release ones.

### Crash Report

It didn't generate a report just a report_000000000960f592.binv (git won't let me attach it)

### System Firmware Version

14.0.0

### Environment?

- Fusee and Hekate
- Official pre-release
- Do you have additional kips or sysmodules you're loading: unknown
- Homebrew software installed: none at launch besides sysdvr
- EmuMMC or SysNAND: unknown

  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?
[image]

Masamune3210, 2022-03-24:
Zip the report and upload it, it's needed to see what is going on

SasorizaTsugare, 2022-03-24:
[fatal_errors.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8337465/fatal_errors.zip)
everything in my fatal errors folder It seems to generate a new one with each boot attempt

SciresM, 2022-03-24:
Wait for mission control to be updated; not an atmosphere issue.

closed 2022-03-24 by SciresM

qvazzler, 2022-03-25:
I can confirm this from my side as well, removing Mission Control fixed the problem above for me.

Since Mission Control isn't updated for 14 yet, so I uninstalled it safely by:

1. Booting into Hekate and launching SD tool, connect it to my PC.
2. Going to https://github.com/ndeadly/MissionControl/releases and downloading 0.6.4
3. Looking at the files inside the zip, then deleting those same files on the switch SD card.
4. Booting up the switch again, now it works.