Atmosphere-NX/Atmosphere #1808: Running Atmosphere

winits, 2022-03-24:
## Bug Report

### What's the issue you encountered?

I can't get Atmosphere to boot from Hekate on an unpatched console, either on sysMMC or emuMMC (partition based) the 2 errors I encounter are "fatal error while running fusee! Invalid GPT signature!" & "fatal error while running fusee! Incorrect version of fusee!"

### How can the issue be reproduced?

I can reproduce the issue by 

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

14.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: Official release version 5.7.0

- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
Partition-based EmuMMC 

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

empithree8, 2022-03-24:
Did you change the payload in the bootloader payload folder reflecting the new fusee.bin that comes with 1.3.0?

SciresM, 2022-03-24:
You need to actually inject the latest version of fusee distributed alongside the relevant update.

Nofbar, 2022-03-24:
How can I reinject payload if I am stuck at "Press POWER to reboot". Do I need to wait till switch is out of power?

takoyaki99, 2022-03-24:
Update your build of Atmosphere, they fixed that. Power off the system completely, then inject your payload once in RCM.

Nofbar, 2022-03-24:
How to Power off if the switch keeps rebooting no matter how long I push the power button?

SciresM, 2022-03-24:
> How to Power off if the switch keeps rebooting no matter how long I push the power button?

1) don't take over other people's issues
2) just hold it for like 30 seconds.

winits, 2022-03-24:
> Did you change the payload in the bootloader payload folder reflecting the new fusee.bin that comes with 1.3.0?

Yes I unzipped the 1.3.0 download in a temp folder on my desktop, then copied the content to my sd card,  confirming file replacement then placing the fusee.bin in the bootloader/payload folder and injected the hekate 5.7.0 payload

closed 2022-03-24 by SciresM