Atmosphere-NX/Atmosphere #1810: A Fatal Error Occurred When Running Atmosphere. 010000000000bd00

CarterBugg, 2022-03-24:
Every time I try to boot into Atmosphere using Fusee or Hekate, I get an error. I'm 100% sure that I'm using the newest versions of everything. (my switch firmware is 14.0.0). Anybody know a solution so that I can use Atmosphere again? Thanks..

666Dudu666, 2022-03-24:
At this moment, i believe, there is no solution with that. I tried for the
last 24 hrs with different setups and i have the same problem. We just have
to be patient

SciresM, 2022-03-24:
1) You didn't use the issue templates
2) Wait for Mission Control to update or remove it, this isn't an atmosphere problem.

closed 2022-03-24 by SciresM

CarterBugg, 2022-03-24:
brain too smooth to use issue templates =)

SciresM, 2022-03-24:
Any issue not using the issue templates will not receive support.