Atmosphere-NX/Atmosphere #1813: Daybreak EmuMMC update Error 0x00007889

Kingslayer9988, 2022-03-25:
## Bug Report

### What's the issue you encountered?

I was updating my EmuMMC to HOS Version 14.0 in Daybreak. I Used Dumped Firmware from SysMMC with runs HOS 14.0 | AMS 13.0 - it does chekc the firmware and says it is vailidated. But upon install I get: </br>
0% complete </br>
Beginning update setup... </br>
Failes to setup update. </br>
Result: 0x00007889  </br>
I did not make any changes to atmosphere just istalled vers. 13.0 and Hekate 5.7.2 </br>

### How can the issue be reproduced?

Install Hekate 5.7.2 and Atmosphere 13.0 update SysMMC with System-Settings updater to HOS 14.0 - Dump Firmware with the new updated Tegraexplorer. After that boot into EmuMMC and start daybreak and install the dumped HOS 14.0 Firmware</br>

### Crash Report

No Crash</br>

### System Firmware Version

13.2.1 | EmuMMC - 14.0.0 | SysMMC </br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate 5.7.2
- Official release or unofficial build:  Offical Release
  - 5.7.2
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: everthing that comes with Atmosphere + Goldleaf
- EmuMMC or SysNAND:
  -EmuMMC, partition-based

### Additional context?

- Additional info about your environment: I am using new sigpatches
[image]

SciresM, 2022-03-27:
This indicates that you have a pending system update; a system update cannot be installed by daybreak when one is pending.

Boot into maintenance mode and reboot to clear pending sysupdates (or just apply the pending sysupdate).

closed 2022-03-27 by SciresM

putnam, 2022-04-14:
Bumping since I repro this on a 10.2 install upgrading to 14.1.0, but I wiped the pending sys update in maintenance mode and am still getting the same error. I am on the latest atmosphere release. Any other suggestions?

putnam, 2022-04-14:
Update to my situation. I restarted several times and got the same error. WiFi was unpaired, so not online, and no pending system update. However I saw a mention elsewhere to turn on Airplane Mode, and somehow that solved the problem. Do we know the technical reason why?

themacintoshnerd, 2022-06-08:
> Update to my situation. I restarted several times and got the same error. WiFi was unpaired, so not online, and no pending system update. However I saw a mention elsewhere to turn on Airplane Mode, and somehow that solved the problem. Do we know the technical reason why?

This worked for me as well rather odd if I do say so myself

dwheeler52, 2022-06-21:
Going into Airplane mode also allowed me to update after the same error code above.

naiij, 2022-08-02:
> Update to my situation. I restarted several times and got the same error. WiFi was unpaired, so not online, and no pending system update. However I saw a mention elsewhere to turn on Airplane Mode, and somehow that solved the problem. Do we know the technical reason why?

Thanks for posting this! It worked for me

tony-aptx4869, 2022-10-20:
> 

Turning on Airplane Mode worked for me too. But still don't know why. So weird!

tony-aptx4869, 2022-10-20:
> Update to my situation. I restarted several times and got the same error. WiFi was unpaired, so not online, and no pending system update. However I saw a mention elsewhere to turn on Airplane Mode, and somehow that solved the problem. Do we know the technical reason why?

Turning on Airplane Mode worked for me too. But still don't know why. So weird!

Wings168, 2022-11-19:
> 更新我的情況。我重新啟動了幾次並得到了同樣的錯誤。WiFi 未配對，因此不在線，也沒有待處理的系統更新。但是我在其他地方看到了打開飛行模式的提示，不知何故解決了問題。我們知道技術原因嗎？

Thanks for posting this! It worked for me too!

Ixolaro, 2023-10-12:
> Update to my situation. I restarted several times and got the same error. WiFi was unpaired, so not online, and no pending system update. However I saw a mention elsewhere to turn on Airplane Mode, and somehow that solved the problem. Do we know the technical reason why?

worked for me to update to 16.1.0

lorenc-tomasz, 2025-05-11:
I was trying to update FW to 20.0.1 and got 2137-8007, so they banned my console even for FW update. Today I tried to update FW using Daybreak and then Error 0x00007889. 

Thanks to this issue I have turned on Flight Mode and it resolved the Error 0x00007889. Now I can confirm that 20.0.1 works with pre-19.0.0.