Atmosphere-NX/Atmosphere #1815: Crashes after 5 seconds of booting

Tenacity2s, 2022-03-26:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

I was restarting my sd card because it would not even boot into atmosphere and now when i boot into it, it boots in atmosphere but after like 5 seconds i get the error screen with error code 2021-0003 (0x615)

### How can the issue be reproduced?

just by booting atmosphere

### Crash Report

Atmosphère Fatal Report (v1.1):
Result:                          0x615 (2021-0003)

Program ID:                      430000000000000b
Process Name:                    sys-botbase
Firmware:                        14.0.0 (Atmosphère 1.3.0-master-b69fa1357)
General Purpose Registers:
Start Address:                   0000000000000000
Stack Trace:
        ReturnAddress[00]:       0000000000000000
        ReturnAddress[01]:       0000000000000000
        ReturnAddress[02]:       0000000000000000
        ReturnAddress[03]:       0000000000000000
        ReturnAddress[04]:       0000000000000000
        ReturnAddress[05]:       0000000000000000
        ReturnAddress[06]:       0000000000000000
        ReturnAddress[07]:       0000000000000000
        ReturnAddress[08]:       0000000000000000
        ReturnAddress[09]:       0000000000000000
        ReturnAddress[10]:       0000000000000000
        ReturnAddress[11]:       0000000000000000
        ReturnAddress[12]:       0000000000000000
        ReturnAddress[13]:       0000000000000000
        ReturnAddress[14]:       0000000000000000
        ReturnAddress[15]:       0000000000000000
        ReturnAddress[16]:       0000000000000000
        ReturnAddress[17]:       0000000000000000
        ReturnAddress[18]:       0000000000000000
        ReturnAddress[19]:       0000000000000000
        ReturnAddress[20]:       0000000000000000

### System Firmware Version

14.0.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Fusee
- Official release or unofficial build: 1.3.0

- Do you have additional kips or sysmodules you're loading:sysbotbase and sigpatches 
- Homebrew software installed: Checkpoint,goldleaf,kosmos tool box,daybreak
- EmuMMC or SysNAND: sysnand
  -

### Additional context?
N/A

SciresM, 2022-03-26:
Remove or update sys-bot base to a version which supports 14.0.0.

closed 2022-03-26 by SciresM

Tenacity2s, 2022-03-26:
god damn you were so certain and guess what you were right thx so much i have been trying to figure this out all day