Atmosphere-NX/Atmosphere #1816: Blue Screen on Mariko / SX Core when booting stock sysnand (package3)

ertaboy356b, 2022-03-26:
### What's the issue you encountered?

Unable to boot stock sysnand (package3) using the latest atmosphere release. I am using Hekate (5.7.2) to boot package3 to stock. No game card was inserted during booting. The firmware update was done via OTA update.

I tried nogc = 0 (previously set to 1) with same results.
I tried deleting the atmosphere folder and replacing it with the updated version with the same results.
No plugin installed whatsoever. No content folder.

Emunand works fine (fusee.bin).

### How can the issue be reproduced?

Boot Mariko with SX Core on sysnand using Hekate with the following settings:
fss0=atmosphere/package3
emummc_force_disable=1
stock=1

### Crash Report

No crash report

### System Firmware Version

14.0.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate fss0=atmosphere/package3
- Official release or unofficial build:  official release version 1.3.0 (pre-release)
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: N/A (emulators)
- EmuMMC or SysNAND: SysNAND

### Additional context?

Hekate's Reboot to OFW works fine.

masagrator, 2022-03-26:
Since this seems to be a copy of issue stated on Discord, I will put the same answer as on Discord.

```
Atmosphere by itself is not allowing to boot into stock using exosphere. It's Hekate that is using those files to boot into stock. Someone would need to prove that this is not a Hekate issue first.
```

SciresM, 2022-03-26:
"stock" not an atmosphere issue.

closed 2022-03-26 by SciresM