Atmosphere-NX/Atmosphere #1818: JKSV saves

jmart8684, 2022-03-27:
## Bug Report

[ Describe the issue in detail and what you were doing beforehand. ]</br> Previously, using JKSV, and earlier, checkpoint, I would be able to make a sav file edit it, load the edited save, and load it in either CFW or Official. For example, to use PKHex. 
With the current pre-release, the edited save would load in CFW, but not in OFW, which was no issue in the past. 

[ Did you make any changes related to Atmosphère itself? ]</br>
Yes

[ If so, make sure to include details relating to what exactly you changed. ]</br>
Updated to pre-release

### How can the issue be reproduced?
Playing/Backing up a save through CFW JKSV,  play on clean OFW. OFW has last save from OFW, CFW loads recent CFW save

Previous release was able to use OFW to load games. Any changes made in CFW wouldn't save to OFW as was able to be done previously

Masamune3210, 2022-03-27:
Sounds like a issue with the homebrew not the system. Are you sure you didn't have save redirection enabled on one profile and not the other or a emunand set up?

SciresM, 2022-03-27:
This really doesn't sound like an atmosphere issue. I'm going to close this and ask you to come talk about details on discord, if you really think this is a cfw problem.

closed 2022-03-27 by SciresM

jmart8684, 2022-03-27:
Also noticed unable to play games on CFW if there's an update required for the game that was updated on official. Using EMUMMC for cfw. Will mess around and look in setting and such to see if I did something different updating to prerelease,  but as far as I know, I set everything up the same as I did before

SciresM, 2022-03-27:
...obviously if you edit save in emummc, it will not affect ofw/sysmmc?