Atmosphere-NX/Atmosphere #1819: error 2016-0247 unable to access microsd card

Leerz, 2022-03-27:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?
error 2016-0247 
unable to access microsd card

[ Describe the issue in detail and what you were doing beforehand. ]</br>
switch lite running ams 1.2.6 trimmed emummc 13,2,1, ofw 13.2, working fine (authentic sandisk class a1 512gb)

new sdcard, 
created new emummc on a new sdcard (trimmed), 
updated to [Atmosphère 1.3.0 (pre-release)]
installed 14.0 with daybreak fat32+xfat
when the console boots up it says error 2016-0247
guaranteed authentic microsd card ( and i get sandisk from official distributor )

[ Did you make any changes related to Atmosphère itself? ]</br>
just updated it, exos for prodinfo and hosts on emummc for ninty servers
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
hekate launches fusee bin 
payload=bootloader/payloads/fusee.bin

- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]
parition based
### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

Masamune3210, 2022-03-27:
Use Fat32, you might not have the exfat driver installed, exfat has colossal issues on the switch anyway

SciresM, 2022-03-27:
Yeah, this is an issue with your card/connector.

closed 2022-03-27 by SciresM

Leerz, 2022-03-28:
hello, yes it is fat32
and it is working fine before updating to fw14 and Atmosphère 1.3.0 (pre-release)

here's how i fixed it, I deleted the EMUMMC folder and it created a new instance. 
normally when i chose not to preserve settings on daybreak, it will prompt me to delete the previous nintendo folder (HOS will prompt on first boot, since the previous nintendo folder was for the previous installation)

it's showing good benchmark speeds on hekate so i know my card is good 
i've replicated this on 2 switches 1 lite and a mariko