Atmosphere-NX/Atmosphere #1820: std::abort() called (0ffe)

dodgypast, 2022-03-27:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

Booting Atmosphere crashes the switch with the error:
std::abort() called (0ffe)

### How can the issue be reproduced?

Mariko Switch with SX mod chip
Was at 13.2.1 HOS with 1.26 Atmospher updated as follows:
Hekate
Atmosphere
Get the latest fusee and put in your payloads directory
Sigpatches
Download 14.0
Install with daybreak

When I select fusee it shows the atmosphere splash screen, then the nintendo spash screen, then crashes.

### Crash Report

https://gist.github.com/dodgypast/726f48d029c6e1bd08cc6fbabe2fbe5e

### System Firmware Version

14.0.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?
Hekate 5.7.2 used to load fussee.bin
SysNAND

Masamune3210, 2022-03-27:
Piracy is not supported

dodgypast, 2022-03-27:
I gotg rid of the contents folder and it fixed so my stupidy.

Apologies for wasting time.

closed 2022-03-27 by dodgypast