Atmosphere-NX/Atmosphere #1825: Error Code: 2001-0123 (0xf601) When installing Atmosphere 1.3 for Switch Update 14.0

JarlTheKing, 2022-03-31:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

### What's the issue you encountered?

Since the Switch updated to 14.0, I needed the latest Atmosphere update (which at this time is 1.3). When installing the update I was given this as a result. I've tried to do some digging on this, seemed like it happed within the past couple years, but I'm still new to this stuff so I don't really know what to do, besides a friend trying to help me out.

[image]

### Crash Report

[01648689446_0100000000000081.log](https://github.com/Atmosphere-NX/Atmosphere/files/8385511/01648689446_0100000000000081.log)

### Environment?

-  Bootloader usedc (fusèe & hekate) was Atmosphère launched by:
- Official release 
- Homebrew software installed: [ * ]
- EmuMMC on SD Card

SciresM, 2022-03-31:
Delete the custom system module. It probably needs to be updated.

closed 2022-03-31 by SciresM

JarlTheKing, 2022-03-31:
> Delete the custom system module. It probably needs to be updated.

Where would I find that, and what would it be called?

SciresM, 2022-03-31:
Delete the atmosphere/contents/0100000000000081/ folder you added.

takoyaki99, 2022-03-31:
In the atmosphere/contents folder delete the one that says 010…..81, and you’ll be good! Make sure to check for updates to that title too, the sysmodule may have been updated for 14.0.

ndeadly, 2022-04-01:
For the record, `0100000000000081` is nx-btred, which was rendered obsolete by the official bluetooth audio support introduced in 13.0.0. It is unlikely to receive further updates and shouldn't be used on firmwares other than 12.x.x.

JarlTheKing, 2022-04-01:
Okay so I deleted the 0100000000000081 Folder, by mistake, without checking what was in the folder itself. Now I am left with this.

[01648770674_0100000000001000.log](https://github.com/Atmosphere-NX/Atmosphere/files/8393845/01648770674_0100000000001000.log)

[image]

takoyaki99, 2022-04-01:
Delete your custom theme using the same process at atmosphere/contents/010….0100.

Masamune3210, 2022-04-01:
Honestly it's probably easier at this point and quicker to just put a clean install of Atmosphere on instead of playing whack-a-mole

JarlTheKing, 2022-04-01:
Just started working, thanks everyone. You really help a brother out, glad you all know the ins & outs of this.