Atmosphere-NX/Atmosphere #1826: fusee.bin doesn't work in RCMLoader device

empithree8, 2022-03-31:
## Bug Report

### What's the issue you encountered?

After renaming current fusee.bin (1.3.0)  to payload.bin, and copy it into the RCMLoader module. Note that it's the only payload.bin in the module. When doing the injection, LED of RCMLoader lights up, but Switch's screen remains blank.

Injection works with fusee.bin (1.2.6), as I still on OFW 13.2.1.

The switch also lost the RCM mode afterwards, needs to press power for 12 seconds to try again. Simply plug in RCMLoader, or try PC injection using TegraRCMGUI will not work immediately.

### How can the issue be reproduced?

Turn off the switch. Plug-in RCMLoader. Power on the switch.

### Crash Report

There's no report generated as the unit doesn't even boot.

### System Firmware Version

13.2.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: RCMLoader module version 1
- Official release or unofficial build: 
  -  Official Pre-released 1.3.0 
- Do you have additional kips or sysmodules you're loading: NIL
- Homebrew software installed: [ NIL ]
- EmuMMC or SysNAND:
   -SysNAND

### Additional context?

- Additional info about your environment:
None that I can think off for now.

pegast73, 2022-03-31:
OMG
1. fuse.bin 1.3.0 is for fw. 14.0.0
2. If you have fw. 13.2.1 so use fuse.bin 1.2.6

Dango1106, 2022-03-31:
ngl i have no idea what this means but I have the same issue can you dumbify it for me

trsludwig, 2022-03-31:
> OMG
> 
> 1. fusee.bin 1.3.0 is for fw. 14.0.0 and ATM 1.3.0
> 2. If you have fw. 13.2.1 so use fusee.bin 1.2.6 and ATM 1.2.6

Sorry, but each version of ATM and fusee.bin brings backwards compatibility. So when a new version of ATM comes out, you can use it even if you are using an older FW. There is one rule: ATM and fusee.bin must be from the same release or at least fusee.bin must be compatible with the version of ATM. Sometimes fusee.bin is not updated, but ATM is.

But why it doesn't work with the RCMLoader has, IMHO, nothing to do with the FW or ATM version used, provided ATM has really been updated to version 1.3.0. Then there should be a different problem. But I am not an expert.

SciresM, 2022-03-31:
The official abbreviation is "AMS", not "ATM".

and anyway this sounds like a user env issue, feel free to come debug on discord.

closed 2022-03-31 by SciresM

NandDixit, 2022-04-01:
https://discord.gg/ZdqEhed

empithree8, 2022-04-05:
@SciresM solved with 1.3.1. Don't know what's the different between 2 fusee.bin. but it is working good now. thanks