Atmosphere-NX/Atmosphere #1828: Fatal error occured when running Atmosphere so I tried loading fusee through tegra instead of hekate

EdgarUstian, 2022-04-01:
## Bug Report

### What's the issue you encountered?

I have updated the atmosphere to the latest version to be compatible with version 14.0.0.
I have replaced the contents of the sc with both the new atmosphere release as well as Hekate
I booted my switch through the Hekate bootloader first and tried running the CFW through the Atmosphere SYS F00 (i don't remember the name)

* This is where the first error occurred.
* Fatal error when running atmosphere

I thought that it was weird and tried changing my bootloader from Hekate 5.7.2 to fusee which came with the latest release of atmosphere.

* This is the second error that I am stuck on now
* A fatal error occurred when running fusee
Switch stuck in a boot loop not being able to parse sdmc:/atmosphere/config/system_setting.ini

Pressing the power button to reboot only brings it back to the same screen

No changes were made to Atmosphere itself

### How can the issue be reproduced?

What I wrote above is how I got to the issue so that is how it can be replicated.

### Crash Report

The most recent crash report I have on my switch is from 3/11/2022 and no new one was generated when having this issue.</br>

### System Firmware Version

14.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- * first Hekate 5.7.2.bin, then the latest version of the fusee.bin
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  * Atmosphère 1.3.0 (pre-release)
- Do you have additional kips or sys modules you're loading:
- Homebrew software installed: [ * ]
*  Not sure. I am just using what atmo is providing me with

### Additional context?

* N/A

Masamune3210, 2022-04-01:
A fatal error occurred when running fusee
Switch stuck in a boot loop not being able to parse sdmc:/atmosphere/config/system_setting.ini

This points to either a malformed system_settings.ini or filesystem corruption. If you haven't changed anything recently, check your SD card for filesystem damage or possibly back everything up to your computer and format the card just to make sure everything is sane

EdgarUstian, 2022-04-01:
How is system_setting.ini supposed to look like? These are the contents for it that I have 

```
button keyword-button scuffle-thema" href="javascript:void(0)" onclick="clickToggle(this);" nx-se-disabled="">
                                                        <div class="button-icon-wrapper">
                                                            <img class="button-icon toggle is-hidden " ref="./check.svg" src="./check.svg" default="is-hidden">
                                                        </div>
```

Masamune3210, 2022-04-01:
Definitely not like that! If you download a fresh copy of the latest release it should have a template for what the file is supposed to look like. That looks like part of a html file or something

EdgarUstian, 2022-04-01:
Lmao that's what I was thinking. But the thing is that I did download a fresh release and moved all the files over. I would think that it would replace some.

But alas, time to fix the setting. Do I just download the regular install and copy-paste the system_setting from one to another?

Masamune3210, 2022-04-01:
Honestly since something has obviously gone wacky it might be better just to back up everything other than the Nintendo Switch folder to your comp and extract a fresh copy anyway. Should get rid of any other errors

Masamune3210, 2022-04-01:
Just saw the edit, glad you got it working! Glad to help

EdgarUstian, 2022-04-01:
I have done that before. Might do it again, but at the moment I am a bit lazy and I just want to play smash bros with mods :^)

But thanks again and I will keep that in mind

closed 2022-04-01 by SciresM