Atmosphere-NX/Atmosphere #1831: A fatal error occurred when running Atmosphère. / Program ID 010000000000bd00 / Error Desc: std::abort() called (0xffe)

qweluke, 2022-04-03:
## Bug Report

### What's the issue you encountered?

Unable to Boot 1.3.0 (pre-release), I just updated it to test on my 14.0 switch firmware. I copied the contents of the release .zip to the root of my sd card and overwritten the conflicting files. I also replaced all of my fusee.bin on both my computer and in my bootloader folder with the one available beside the new atmosphere. Issue occurs with both hekate launch (5.7.2) and tegrarcmgui launch (2.6).
### How can the issue be reproduced?

replace existing atmosphere and fusee.bin files with the new pre-release ones.

### Crash Report

below as an attachement

### System Firmware Version

11.0.0

### Environment?

Fusee and Hekate
[report_000000000590a079.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8405001/report_000000000590a079.bin.zip)

Official pre-release

Do you have additional kips or sysmodules you're loading: unknown

Homebrew software installed: none at launch besides sysdvr

EmuMMC or SysNAND: unknown

[ If using an EmuMMC, include whether it's partition-based or file-based. ]

SciresM, 2022-04-03:
Remove/update mission control.

closed 2022-04-03 by SciresM

qweluke, 2022-04-03:
@SciresM **I don't have missing controll at all!!**