Atmosphere-NX/Atmosphere #1832: NO MISSION CONTROL - A fatal error occurred when running Atmosphère. / Program ID 010041544d530000 / Error Desc: std::abort() called (0xffe)

qweluke, 2022-04-03:
## Bug Report

### What's the issue you encountered?

Unable to Boot 1.3.0 (pre-release), I just updated it to test on my 14.0 switch firmware. I copied the contents of the release .zip to the root of my sd card and overwritten the conflicting files. I also replaced all of my fusee.bin on both my computer and in my bootloader folder with the one available beside the new atmosphere. Issue occurs with both hekate launch (5.7.2) and tegrarcmgui launch (2.6).
### How can the issue be reproduced?

replace existing atmosphere and fusee.bin files with the new pre-release ones.

### Crash Report

[report_000000000590a079.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8405001/report_000000000590a079.bin.zip)

### System Firmware Version

11.0.0

### Environment?

Fusee and Hekate

Official pre-release

Do you have additional kips or sysmodules you're loading: unknown

Homebrew software installed: none at launch besides sysdvr

EmuMMC or SysNAND: unknown

[ If using an EmuMMC, include whether it's partition-based or file-based. ]

SciresM, 2022-04-03:
Ah, I did misread.

Fix archive bits.

SciresM, 2022-04-03:
(report linked really is the usual archive bits problem error report where PathNotFound() is returned trying to access atmosphere/automatic_backups).

SciresM, 2022-04-03:
Also, I didn't misread. You edited the title of your issue to remove the mission control program ID. 

I think you are not faithfully reporting information, though this will resolve on archive bit fix anyway.

closed 2022-04-03 by SciresM

qweluke, 2022-04-03:
@SciresM how to fix archive bits?

SciresM, 2022-04-03:
This issue tracker is for bugs, not for support. Seek support on discord.

That said, run the fixer tool in hekate.