Atmosphere-NX/Atmosphere #1833: Panic ocurred while running Atmosphere - sysMMC and emuMMC os v. 1.14.0 - atm v. 1.3.0

ivo-hr, 2022-04-03:
## Bug Report

### What's the issue you encountered?

Atmosphere refuses to boot in the latest release and on the lastest OS version, showing  the following:
Panic ocurred while running atmosphere
Title ID: 010000000000BD00
Error: std::abort (0xFFE)
I installed 1.14.0 by accident, so I waited for a new atmosphere release and when it happened, I copied the files and pasted them over my SD card, replacing the files in the destination. I removed anything about custom themes as the new os had new elements, but didn't touch nothing else.

### How can the issue be reproduced?

1. Update your Switch to version 1.14.0 before updating Atmosphere
2. Update Atmosphere by inserting the SD card on a PC and pasting the new CFW, replacing files in the destination
3. Update Hekate to v5.7.2
4. Try to boot either emu or sysMMC

### Crash Report

[crash_reports.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8405299/crash_reports.zip)
Here are both .bin, the one ending in 94 is sysMMC and the other one (ending in 91) is emuMMC

### System Firmware Version

1.14.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
  Hekate v5.7.2. Also tried fuseé but to no avail.
- Official release or unofficial build: 
  Latest official release, 1.3.0

- Do you have additional kips or sysmodules you're loading:
  I did but deleted them in case it was the problem.
- Homebrew software installed:
  Skyline and ARCropolis, everything else are simple .nro
- EmuMMC or SysNAND:
  Both, EmuMMC is partition.

### Additional context?

Yeah, I hate Nintendo's update prompt defaulting to "update" and watch as everything crumbles before your eyes, unable to do a thing, while it updates.

SciresM, 2022-04-03:
Remove/update mission control.

closed 2022-04-03 by SciresM

ivo-hr, 2022-04-03:
Removed missioncontrol, same issue.

ivo-hr, 2022-04-03:
> Remove/update mission control.

Did that, now it crashes in the blue Atmosphere crash screen with an error code 2162-0002 (0x4a2).

ivo-hr, 2022-04-03:
> > Remove/update mission control.
> 
> Did that, now it crashes in the blue Atmosphere crash screen with an error code 2162-0002 (0x4a2).

Ok, im an idiot, i thought i deleted the Theme /contents/ folder but i didn't. Thx!