Atmosphere-NX/Atmosphere #1834: Issue when building 1.3.0

ajbritain4, 2022-04-04:
### What's the issue you encountered?

Error while building using devkitpro - all dependencies installed (I think?) 

[image]

### How can the issue be reproduced?

Attempt to clone and build from source.

### Crash Report

N/A

### System Firmware Version

N/A 

### Environment?

1.3.0 Official build, Windows 11, all dependencies as of https://github.com/Atmosphere-NX/Atmosphere/blob/master/docs/building.md installed 

### Additional context?

N/A. Built 1.2.6 successfully a few months back and don't _think_ my config has changed since

SciresM, 2022-04-04:
Use the 14.0.0 support PR to libnx.

closed 2022-04-04 by SciresM

Takokeshi, 2022-04-06:
My build chokes on the new functions even though I replaced libnx with your updated fork. I'm not really sure what else I can do :/ (everything else is otherwise up-to-date.)

<details>
    <summary>Log</summary>

```
linking fatal.elf
/opt/devkitpro/devkitA64/bin/../lib/gcc/aarch64-none-elf/11.2.0/../../../../aarch64-none-elf/bin/ld: /opt/devkitpro/devkitA64/bin/../lib/gcc/aarch64-none-elf/11.2.0/../../../../aarch64-none-elf/bin/ld: DWARF error: can't find .debug_ranges section.
fatal_task_power.o: in function `ams::fatal::srv::(anonymous namespace)::PowerButtonObserveTask::Run()':
fatal_task_power.cpp:(.text._ZN3ams5fatal3srv12_GLOBAL__N_122PowerButtonObserveTask3RunEv+0xf0): undefined reference to `bpcGetPowerButton'
collect2: error: ld returned 1 exit status
make[4]: *** [/opt/devkitpro/libnx/switch_rules:80: /home/user/Atmosphere/stratosphere/fatal/out/nintendo_nx_arm64_armv8a/release/fatal.elf] Error 1
make[3]: *** [/home/user/Atmosphere/stratosphere/fatal/system_module.mk:71: all] Error 2
make[2]: *** [/home/user/Atmosphere/stratosphere/stratosphere.mk:14: fatal] Error 2
make[1]: *** [/home/user/Atmosphere/atmosphere.mk:127: stratosphere] Error 2
make[1]: Leaving directory '/home/user/Atmosphere'
make: *** [Makefile:39: nx_release] Error 2
```

</details>