Atmosphere-NX/Atmosphere #1838: Error when opening album on FW 14.1.0 Atmosphère 1.3.1

jlavoieqc, 2022-04-05:
## Bug Report

Launching hbmenu cause atmosphère crash. Latest hbmenu used and latest hbloader used

### What's the issue you encountered?

When I launch hbmenu with album, atmosphère crashes. It make the same thing when holding r button when launching a game (The game crashes)

### How can the issue be reproduced?

Opening album

### Crash Report

https://gist.github.com/jlavoieqc/01b58fd7ed0b65d78e60ef7a10ccfaa7

### System Firmware Version

FW 14.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by : hekate
- Official release
  - Official release version 14.1.0
- no
- Daybreak, ChoiDuJourNX, GoldLeaf, appstore
- SysNAND

### Additional context?

N/A

UltraSyko, 2022-04-08:
Can confirm, this happens. However, you can safely enter hbmenu with no game running. So just make sure you close your game if you really need to open it.

SciresM, 2022-04-08:
1) works fine for me, this isn't fundamentally broken at least.

2) the error report you posted described is from hbloader trying to load an NRO:

```c
    if (read(fd, header, sizeof(*header)) != sizeof(*header))
        diagAbortWithResult(MAKERESULT(Module_HomebrewLoader, 4));

    if (header->magic != NROHEADER_MAGIC)
        diagAbortWithResult(MAKERESULT(Module_HomebrewLoader, 5));
```

It sounds like your hbmenu.nro is corrupt. Try replacing it with the one from the release zip/check for SD corruption.

Closing because this seems like a user env issue and not a bug, per se.

closed 2022-04-08 by SciresM