Atmosphere-NX/Atmosphere #1841: Holding L while booting a game does not disable IPS Patches

Mars2032, 2022-04-07:
## Bug Report

### What's the issue you encountered?

While trying to boot a game that normally as mods installed via both the contents folder and exefs_patches folders, only the contents folders will be disabled. For games that use an IPS patch for external code, this causes the game to crash, as the external code portion in contents is not enabled, but the IPS patch still is. Manually renaming the IPS patch to be invalid fixes the crash.

### How can the issue be reproduced?

Install any mod for any game that has a .ips file and a file named subsdk1, and boot it while holding L. 

### Crash Report
[01649379419_0100000000010000.log](https://github.com/Atmosphere-NX/Atmosphere/files/8447112/01649379419_0100000000010000.log)

### System Firmware Version

13.2.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
  - Hekate
- Official release or unofficial build: 
  - 5.7.0
- Do you have additional kips or sysmodules you're loading:
  - N/A
- Homebrew software installed: 
  - Goldleaf v0.9
  - sys-ftpd-light
  - IPSWitch
- EmuMMC or SysNAND:
  - SysNAND

### Additional context?

- Additional info about your environment:
  - I mainly mod Super Mario Odyssey, so for some reason this could be game specific, but based on how the crash works, I do not think it is.

SciresM, 2022-04-07:
"Use an ips patch for external code"...what?

Mars2032, 2022-04-07:
> "Use an ips patch for external code"...what?

I may have just worded it poorly. What I mean is using ips patches to place branch instructions to the subsdk1 in the original code.

SciresM, 2022-04-07:
That is...not really how stuff is designed to be used. I can think on it, there are some really big problems with keypress controlled patching though.

yourboid, 2022-05-14:
[image]

ruje0504, 2022-09-02:
Therefore, this kind of IPS generated for mod should be done as little as possible, and it is more stable to directly modify exefs

closed 2022-12-23 by SciresM