Atmosphere-NX/Atmosphere #1842: Panic occurred while running atmosphere when resetting games repeatedly

Elvis0077, 2022-04-08:
## Bug Report

### What's the issue you encountered?

When resetting Pokemon Legends Arceus/Pokemon Sword ~250 times (both digital copies from E-shop), an error pops up that says "The software was closed because an error occurred"

Then, when attempting to launch any game (not just Pokemon) from the HOME Menu, an error will pop up saying "Unable to start software.  Return to the HOME Menu and try again"

Finally, when launching the Homebrew menu from the photo album in order to reboot to payload, the following crash happens:

"Panic occurred while running Atmosphere.
Title ID:  0100000000000002
Error:  DABRT <0x101>"

I removed ldn_mitm and was resetting Legends Arceus with just SBB and still got a crash.  I only put ldn_mitm back on the Switch to attempt resets on a different game (Sword).

I have made no changes to Atmosphere itself.

### How can the issue be reproduced?

As mentioned in the description above, simply resetting a game repeatedly (average 250 times) causes the software error and then launching HB menu from album causes the crash

### Crash Report

All crash reports I gathered are in the zip file below.  I was not sure how to create a gist but am happy to do so if you need the crash reports in a different way

[Crash Reports.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8447835/Crash.Reports.zip)

### System Firmware Version

14.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:  **Hekate 5.7.2**
- Official release or unofficial build: Official Release **1.3.1|S**
- Do you have additional kips or sysmodules you're loading:  Yes; **Sysbot-USB-botbase 2.1.1 hybrid** and **Experimental ldn_mitm for 14.1.0** (ldn_mitm was only used for Sword; Legends Arceus was SBB only)
- Homebrew software installed: Daybreak, **FTPD Pro 3.0.0, HB App Store, JKSV, NX-Shell, NXThemes Installer**
- EmuMMC or SysNAND:  **SysNAND**

### Additional context?

N/A

Sorata, 2022-04-08:
Delete the theme

Elvis0077, 2022-04-08:
I have no themes on either of these Switches

Sorata, 2022-04-08:
OK, delete /atmosphere/contents/

Elvis0077, 2022-04-08:
I can do this and then do manual resets later to see if it still crashes.  The only sys-module it could be is SBB since I had deleted all but that and still got the crash with PLA.  Three other users I know are also having this exact issue.  I average 250, another averages 330 and another wasn't counting but was probably 500+.  It seems something is leaking.  I'll follow-up with the results if I do the manual resets after deleting my contents folder.

SciresM, 2022-04-08:
@Elvis0077 Yes, it looks like there's some kind of memory leak. Good spot, this'll take some effort to investigate though.

fwiw, your report comes from nullptr being returned by new.

closed 2022-04-08 by SciresM

SciresM, 2022-04-08:
Fixed, this will make it into Atmosphere 1.3.2 in a few days.

In the meantime, if this is a big issue for you, you can use this hotfix zip (I won't be updating the official 1.3.1 release though):

[atmosphere-1.3.1-master-106599895+hbl-2.4.1+hbmenu-3.5.0.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8454341/atmosphere-1.3.1-master-106599895%2Bhbl-2.4.1%2Bhbmenu-3.5.0.zip)
[fusee-106599895.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8454346/fusee-106599895.zip)

Thank you for finding this problem!

commit 10659989 (2022-04-08): fs: fix memory leak when path is reallocated (closes #1842)