Atmosphere-NX/Atmosphere #1844: crash after updating to 14.1.0

mrvnklm, 2022-04-08:
## Bug Report

After I updated my switch using daybreak from 13.0.0 to 14.1.0 the system crashed on reboot.

### What's the issue you encountered?

I tried resetting the switch completely and updating manually / via TegraExplorer, without any success. I did use a fresh SD card (FAT32 and exFAT) filled with latest AMS release (1.3.1) and nothing else. I used fusee to boot the switch, without the use of hekate or any other software.

### Crash Report

https://gist.github.com/mrvnklm/28adf2de543d1d13e6496d22562bd63a

### System Firmware Version

14.1.0

### Environment?

clean AMS 1.3.1 + fuess, no sysmodules or anything else

### Additional context?

The switch I am using has been recovered by replacing the broken EMMC with a new one, so as there was no backup the PRODINFO is from a different console / wiped and therefore the serial number is X1000...

lsp199308, 2022-04-08:
https://github.com/CaramelDunes/prodinfo_gen/issues/24

SciresM, 2022-04-08:
Yeah, this isn't an atmosphere issue. Seek support with whatever you're using to generate mmc.

closed 2022-04-08 by SciresM

marcselman, 2022-05-04:
@mrvnklm I'm sorry to do it this way, I could find no other way of contacting you. Quick question: what happened to your homeassistant-desktop project? Is it moved or did you stop working on it?