Atmosphere-NX/Atmosphere #1849: Template test

beyondmeat, 2022-04-13:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

beyondmeat, 2022-04-13:
@SciresM you should switch the help info to comments then they don't appear when you submit a PR

SciresM, 2022-04-13:
@beyondmeat what?

I don't understand why you made a an empty template issue. It seems a little rude/time wasting to do so without messaging me on discord ahead of time.

closed 2022-04-13 by SciresM

beyondmeat, 2022-04-13:
@SciresM all the template help text you have between the square brackets should be changed so they are markdown comments instead. That way none of that help text that clutters up the template is visible (remains visible in edit mode). This will make the issues more readable while achieving the same thing. The result is a cleaner filled out template.

For extra clarification: I'm talking about the template help like ```[ If using an EmuMMC, include whether it's partition-based or file-based. ]``` if you change that to be a comment, when the issue is submitted the comments are not visible. This is actually a common practice when using the template feature.

Not everyone is on discord and I'm sorry you found this rude. I just noticed a lot of people aren't filling out your templates and that got me looking at them.

beyondmeat, 2022-04-13:
Also the biggest reason is so users no longer need to remove the lines before positing @SciresM

SciresM, 2022-04-13:
@beyondmeat the whole point of the template is to make the user go through the process of editing it all.

Lazy users who submit issues without fully replacing all the text will have their issues closed and be told to reopen with the template filled out. 

The way it works out maximizes my ability to tell when someone hasn't filled it out quickly, which saves time and energy.

I do not believe that what you describe will result in filled out templates. I believe it will result in less complete templates.

It will eventually be replaced by the GitHub forms which make filling out the fields actually required, but that's a low priority TODO item.

For what it's worth, development happens on discord. If you're interested, message SciresM#0524 -- an issue like this without a message to me is just confusing. 

My intent would have been for you to click the "question" template which tells you to use discord.