Atmosphere-NX/Atmosphere #1850: A fatal error ocurred when running Atmosphere

ReceptiveRaptor, 2022-04-14:
a fatal error occured when running atmosphere
Program ID: 0100000000000005
Error Desc: std::abort() called (0xffe)
report saved fatal_errors/report.bin
Press POWER to reboot.

What's the issue you encountered?
I updated atmosphere, hekate and sigpatches, and attempted to launch into emuMMC and atmosphere crashes.

How can the issue be reproduced?
I haven't tried reproducing it in another environment but this crash happens every time no matter what I do.

Crash Report
[report_0000000000a52514.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8492781/report_0000000000a52514.zip)

System Firmware Version
switch firmware 13, atmosphere 1.3.1

Environment?
What bootloader (fusèe, hekate, etc) was Atmosphère launched by: TegraSmashGUI.exe
Official release or unofficial build: Official release 1.3.1
Do you have additional kips or sysmodules you're loading: No
Homebrew software installed: [Appstore, EdiZon, JKSV,Retroarch,aio installer, awoo-installer, checkpoint, DBI, SkyNX, tinfoil, ftpd, hbl]

Additional context?

I was able to boot atmosphere in cfw sysMMC but I don't use sysMMC I use emuMMC
I also tried this fix:
Get your switch in RCM mode.
Inject Hekate payload
Tap on "Tools" on top of the hekate screen.
Tap on "Archive Bit - AutoRCM" on the right bottom.
Tap on "Unset archive bit" button. This might take a while, be patient.
and got 0 set/unset

SciresM, 2022-04-14:
Piracy is not supported.

closed 2022-04-14 by SciresM

ReceptiveRaptor, 2022-04-14:
Cool, but that doesn't say what the issue is?

SciresM, 2022-04-14:
It does not, correct -- support is not being provided due to your engaging in piracy.