Atmosphere-NX/Atmosphere #1851: 14.1.0 cant open anything

bold-sama, 2022-04-15:
## Bug Report

### What's the issue you encountered?
after updating HOS, atmosphere and hekate i can no longer launch applications or launch into homebrew. i checked my flags in hekate and everythings fine, how can i check my sig patches? or is there another problem? i cant even open homebrew menu, ive included a recent crash report of when i tried to open hbmenu applet mode in album.
### How can the issue be reproduced?
this is a fresh update that cant boot app

### Crash Report
https://gist.github.com/bold-sama/0a6729a1d8a8196a6ad73a35b98d493f

### System Firmware Version

14.1.0</br>

SciresM, 2022-04-15:
Piracy is not supported; if games won't open, try redownloading them from the eShop.

closed 2022-04-15 by SciresM

bold-sama, 2022-04-15:
Except this switch is banned, I can’t open Eshop, my concern isn’t piracy. I physically can’t open HBmenu and that’s my concern.

bold-sama, 2022-04-15:
> Piracy is not supported; if games won't open, try redownloading them from the eShop.

Except this switch is banned, I can’t open Eshop, my concern isn’t piracy. I physically can’t open HBmenu and that’s my concern.