Atmosphere-NX/Atmosphere #1852: Can not open homebrew

bold-sama, 2022-04-15:
## Bug Report

I can not launch into homebrew. Using R and opening an application gives me a black screen and when I use the album mode it crashes. This is a fresh update too 14.1.0 I haven’t done anything special too the switch besides remove folders and do a fresh install. 

### Crash Report

https://gist.github.com/bold-sama/0a6729a1d8a8196a6ad73a35b98d493f

### System Firmware Version

14.1.0

### Environment?
Hekate launching fuss

Masamune3210, 2022-04-15:
Please go back and fill out the rest of the template

bold-sama, 2022-04-15:
> Please go back and fill out the rest of the template

I did? Everything else was N/A. What am I missing

SciresM, 2022-04-15:
Support is not provided for piracy.

closed 2022-04-15 by SciresM

bold-sama, 2022-04-15:
> Support is not provided for piracy.

Where is this piracy coming from??? I am just trying to figure out with I can’t launch my homebrew. I’m not pirating anything. Like the applet menu won’t open. What does this have to do with piracy.

SciresM, 2022-04-15:
Your previous issue indicated games not opening and explicitly mentioned piracy patches.

Engaging in piracy --> support not provided.

Also, the issue tracker is for bugs, not for support anyway.

You should be getting support on discord, not here (although RS also does not provide support in cases of piracy).

bold-sama, 2022-04-15:
> Your previous issue indicated games not opening and explicitly mentioned piracy patches.
>
> Engaging in piracy --> support not provided.
>
> Also, the issue tracker is for bugs, not for support anyway. [...]
  Are you talking about sig patches? I can move too discord for help but im not pirating anything. Im literally trying to get my switch too open everything again. I can’t open homebrew menu or launch anything. I just want to figure out why I can’t launch homebrew.

SciresM, 2022-04-15:
Signature patches = piracy.

Their use or mention = support not provided, sorry.