Atmosphere-NX/Atmosphere #1859: Latest build (1.3.1) supports latest FW release (14.1.1) - Update release page

ivo-hr, 2022-04-20:
Just saying that the latest 14.1.1 update appears to be supported by Atmosphere 1.3.1, as I updated my console by mistake and it still runs Atmosphere well.
IDK if there's much behind-the-scenes changes, but if there aren't the release page should be updated to reflect this info.

Thanks for all the hard work Scires!

masagrator, 2022-04-20:
Issues are not for saying that something is working, but saying what is not working.

ivo-hr, 2022-04-20:
Yeah, but i don't know where to put it so i guessed here would be the first place anybody would look to see if there's any issues with the latest update.
if there's another place where I could post this, please tell me.
Thanks!

masagrator, 2022-04-20:
Just check SciresM twitter next time

https://twitter.com/SciresM/status/1516211106267623425

SciresM, 2022-04-20:
The release page doesn't list the maximum supported firmware version explicitly, so far as I'm aware.

closed 2022-04-20 by SciresM