Atmosphere-NX/Atmosphere #1862: Installing or Uninstalling Files causes problems until restart

WarioK, 2022-04-21:
## Bug Report

### What's the issue you encountered?

Example issue: going to Settings > Data MAnagement and deleting software will cause all lots of software to start reporting file size of 0.0MB. At this point, trying to launch any software will result in error or system freeze.  Restarting atmosphere then fixes the issue.

### How can the issue be reproduced?

### System Firmware Version

14.1.1 and 14.1.0
### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee and hekate, tried both
- Official release or unofficial build: 
Official release 1.3.0
- EmuMMC or SysNAND:
SysNAND

### Additional context?

- Additional info about your environment:
software is installed to 1TB Micro SD card formatted Fat32

masagrator, 2022-04-21:
Update Atmosphere to 1.3.1 and check if issue still persists.
You should use latest available Atmosphere for reporting issues.

monstrosities, 2022-04-21:
Atmosphere 1.3.0 can't run firmware 14.1.0+

WarioK, 2022-04-21:
I'm sorry, that was a misstype on my part.  I just verified I am running Atmosphere 1.3.1

JCamG, 2022-04-22:
Can confirm, AMS 1.3.1 and firmware 14.1.0, i have to restart the Switch for the games to work again, otherwise they give error at launch and at data management they appear as 0.0mb.

lsp199308, 2022-04-22:
https://github.com/Atmosphere-NX/Atmosphere/issues/1842

Deathreed, 2022-04-22:
Good thing I saw this because I also have this issue and I thought my switch or even my card is the problem.

daveftw84, 2022-04-22:
happens to me too. i reformatted my card because of it

cherubyn, 2022-04-22:
I had the same issue (but only once so far). AMS 1.3.1, Firmware 14.1.0. Tried to reproduce the issue without "success" so I thought this was related to a network glitch/checksum error whatever...

cucholix, 2022-04-24:
https://github.com/Atmosphere-NX/Atmosphere/issues/1862#issuecomment-1105901337
This fixed the installing/uninstalling error for me.

daveftw84, 2022-04-24:
> #1862 (comment)
> This fixed the installing/uninstalling error for me.

sweeeeet

WarioK, 2022-04-24:
> [#1862 (comment)](https://github.com/Atmosphere-NX/Atmosphere/issues/1862#issuecomment-1105901337) This fixed the installing/uninstalling error for me.

I'm cautiously optimistic that it did for me as well!  Haven't had the issue since trying this hotfix.  Had another more minor issue in its place, but I'll wait for the full release to see if that's resolved as well

Dsseed, 2022-04-25:
This bug has always existed, and there is no official update, only the fix can be found in the problem

WarioK, 2022-04-25:
> This bug has always existed, and there is no official update, only the fix can be found in the problem

I don't know mate, I've been using Switch and Atmosphere for years and just now started encountering it

SciresM, 2022-04-25:
It sounds like this is possibly the memory leak I fixed in path handling a few weeks ago. If this is triggered using the hotfix build linked above, please let me know.

closed 2022-04-25 by SciresM