Atmosphere-NX/Atmosphere #1863: A fatal error occurred when running Atmosphere

KamiSquadX, 2022-04-22:
## Bug Report
"A fatal error occurred when running Atmosphere.
Program ID: 0100000000000005
Error Desc: std::abort() called (0xffe)

Report saved to /atmosphere/fatal_errors/report_000000000ccad89.bin"

### What's the issue you encountered?

After booting up Atmosphere, through Hekate 5.7.2 and fusee 1.3.1, via payload injector dongle multiple times, screen eventually goes black and I receive the fatal error message

### How can the issue be reproduced?

Boot to RCM
Launch Hekate
Boot Atmosphere through either fusèe-primary or emunand

### Crash Report

[report_0000000000ccad89.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8536675/report_0000000000ccad89.zip)

### System Firmware Version

14.1.1</br>

### Environment?

EmuMMC
Partition-based

PixLAK, 2022-04-25:
having the same issue with different program ID. only happens when I attempt to load a second player profile while playing diablo 3.

sonicfan2007, 2022-04-28:
I've got the exact same issue. Same situation.

sonicfan2007, 2022-05-02:
Also, updating to 1.3.2 doesn't resolve the issue.

ENjoyBlue2021, 2022-05-05:
I also have the same Problem since I updated to 14.0+ with newer athmosphere version.
I can play for around 30-60 min but eventually the game crashes or I get the athmosphere error screen.
Didnt happen before. I'm also using partition EmuMMC.

yourboid, 2022-05-14:
[image]

BaptisteRo, 2022-08-24:
> I fixed my issue in a simple way - go to atmosphere/contents and delete the folder with the titleid from the error. Do the same if it happens again.

Thanks man i love you

Albanmak, 2022-11-27:
i don't see  0100000000000005 file

Masamune3210, 2022-11-27:
Not every appearance of this issue is from some weird off the block mod stuffed into the contents folder, if it doesn't apply to you, please go make a new issue and fill out the issue template

Masamune3210, 2022-11-27:
> having the same issue with different program ID. only happens when I attempt to load a second player profile while playing diablo 3.

if its a different titleid, please make a new issue and fill out the template. Combining multiple issues into one report like this is not the best way to find your issue solved.

Masamune3210, 2022-11-27:
> I also have the same Problem since I updated to 14.0+ with newer athmosphere version. I can play for around 30-60 min but eventually the game crashes or I get the athmosphere error screen. Didnt happen before. I'm also using partition EmuMMC.

same for you, if the titleid is different, please make a new issue and fill out the issue template

closed 2022-12-23 by SciresM