Atmosphere-NX/Atmosphere #1864: Crash on boot (2105-0602)

NamelessGhouls, 2022-04-23:
## Bug Report

### What's the issue you encountered?

Every time I start the console after being in the dock and I try to access configuration or any element of the UI, the console crashes giving me the failure that I will attach later. When I try to turn it back on, Atmoshpere does not start and the screen stays black. I have tried to make clean installations of Atmoshpere formatting the partitions( I tried in exfat and fat32) and the first time it runs it works but if I turn it off and turn it on again it crashes again.

I have not changed anything in Atmoshpere.

### How can the issue be reproduced?

-I turn on the console after being in sleep mode (either in the dock or out of the dock) 
-Try to enter an ui options (ex: Controllers)
-The console freezes for a moment and crashes.

### Crash Report

[01641020112_010000000000002f.log](https://github.com/Atmosphere-NX/Atmosphere/files/8548247/01641020112_010000000000002f.log)

### System Firmware Version

14.1.0

### Environment?

- Fusee
- Official release version 1.3.1
- I don't have any sysmodules or kips
- Homebrew software installed: Daybreak, reboot to payload
- Using EmuMMC partition-based.

yourboid, 2022-05-14:
[image]

leopitto92, 2022-07-21:
Did you solve it?
I'm experiencing the same issue...

Dokman, 2022-07-24:
> Did you solve it? I'm experiencing the same issue...

Me too

Dokman, 2022-08-07:
> Did you solve it? I'm experiencing the same issue...

Redo your EmuMMC

closed 2022-12-23 by SciresM