Atmosphere-NX/Atmosphere #1865: emuMMC doesn't have access to sd card

BaamAlex, 2022-04-24:
## Bug Report

### What's the issue you encountered?

I created a new emuMMC today with hekate. But every time i try to launch it, i get the error 2016-0247 after factory resetting in the home menu. The funny thing is, the ofw is pretty fine. The ofw has normal access to the sd card without any issues. But i dunno if it is atmosphere related or a problem with my sd card.

Did you make any changes related to Atmosphère itself? 

No.

### How can the issue be reproduced?

Factory resetting the emuMMC and try to get into the home menu.

### Crash Report

I don't get an error from atmosphere itself. It's the hos which gives me the error. 

### System Firmware Version

14.1.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Atmosphère 1.3.1+Hekate 5.7.2 ]

- EmuMMC or SysNAND:
  - [ emuMMC partition-based. Same thing happens in a file-based emuMMC. ]

lsp199308, 2022-04-25:
https://en-americas-support.nintendo.com/app/answers/detail/a_id/22720/p/897

shchmue, 2022-04-25:
This is in the NS SD result range, it's to do with being unable to properly conduct SD seed matching and suggests there's something wrong with the card itself or with the setup of the emummc.

BaamAlex, 2022-04-25:
The error is completely new for me. So...does that mean the sd card is dying? Should I buy a new one?

shchmue, 2022-04-25:
that would be a good troubleshooting step, i'm skeptical this has anything to do with CFW and you might want to seek troubleshooting in an appropriate discord server

BaamAlex, 2022-04-25:
Which discord server? The reswitched one?

morterox, 2022-04-28:
I had the same problem when updating with AIO updater, I had my SD card in exFAT so I thought that could be the reason. I backed up all my save games with Edizon (because checkpoint wasn't working for me) and created a new emuMMC with FAT32, my SD card is working great and no sign of a problem with it for the last 2 weeks. This was obviously a bug but I couldn't sort it out.

BaamAlex, 2022-04-29:
I don't use neither the aio Updater nor exFAT.

ELY3M, 2022-04-29:
I think we need to check sd reader to see if it is issue.  it can be missing pins in the connector part.  have to use hekate payload and do testing.    read more here https://gbatemp.net/threads/1-bit-mode-on-hekate.587690/

BaamAlex, 2022-04-29:
No. Hekate doesn't show me the 1-bit mode.

closed 2022-04-30 by SciresM

BaamAlex, 2022-05-04:
I got my new sd card today. And the same error appears with the new sd card. And as i said, hekate doesn't show me the 1-bit mode. I dunno why this were closed, but the problem is still present.

morterox, 2022-05-04:
@BaamAlex what happened to me is that there was no way of saving the emuMMC that seemed to be the problem, after recreating everything I have been playing for almost 1 month without an issue

BaamAlex, 2022-05-04:
I know that the old files will be deleted. But that the sd isn't accessible after a factory reset, is completely new for me. I teardown my console now.