Atmosphere-NX/Atmosphere #1869: Error after update 14.1.1

Poop81, 2022-04-30:
hi, i want to ask for your help pls, after updating the switch to 14.1.1, this one crashes on a screen, see photo below.

Before the update, I was on FW 13.0.0 and AMS 1.2.5.
I started by doing the update with daybreak of the console before wanting to update AMS. 

Before that, I had no theme installed, and to update,
I removed all my files to put the new ones from AMS and Hekate.

I left only the emuMMC folder and it still doesn't work.
Can you help me to solve the problem pls . THX

[image]

Masamune3210, 2022-04-30:
Please go back and fill out the issue template correctly

SciresM, 2022-05-01:
Reopen using the issue templates.

closed 2022-05-01 by SciresM