Atmosphere-NX/Atmosphere #1870: Crash Error Code (2105-0584) after update 14.1.1 #1869

Poop81, 2022-05-01:
## Rapport d'erreur

[ Si une section ne s'applique pas, remplacez son contenu par "N/A". ]</br>
[ Les lignes entre [ ] (crochets) doivent être supprimées avant la publication. ]</br>

[ * ]</br>
[Remarque : Si le bogue ou le plantage que vous avez rencontré est lié ; ]</br>
[ - logiciel utilisé pour faire des "sauvegardes", ]</br>
[ - logiciels distribués explicitement pour le piratage, etc ] </br>
[ alors les contributeurs ne fourniront pas de support pour votre problème et votre problème sera fermé. ]</br>

### Quel est le problème que vous avez rencontré ?

[ salut, je veux demander votre aide svp, après la mise à jour du commutateur vers 14.1.1, celui-ci se bloque sur un écran, voir la photo ci-dessous.

Avant la mise à jour, j'étais sur FW 13.0.0 et AMS 1.2.5.
J'ai commencé par faire la mise à jour avec daybreak de la console avant de vouloir mettre à jour AMS.

Avant cela, je n'avais aucun thème installé, et pour mettre à jour,
j'ai supprimé tous mes fichiers pour mettre les nouveaux d'AMS et Hekate.

Je n'ai laissé que le dossier emuMMC et cela ne fonctionne toujours pas.
Pouvez-vous m'aider à résoudre le problème svp. THX
 ]</br>
[ Avez-vous apporté des modifications liées à Atmosphère elle-même ? ]</br>
[ Si c'est le cas, assurez-vous d'inclure des détails concernant exactement ce que vous avez modifié. ]</br>

### Comment le problème peut-il être reproduit ?

[ * ]</br>
[ Incluez un processus détaillé, étape par étape, pour recréer votre problème. ]</br>

### Rapport d'incident

[ Les rapports de crash peuvent être trouvés sous ``/atmosphere/crash_reports``. ]</br>
[ Si votre problème a causé le plantage d'Atmosphère, incluez le ou les rapports de plantage en créant un [gist](https://gist.github.com/) et en collant le lien ici. ]</br>
[ Si vous n'incluez pas de rapport de plantage dans les cas de problèmes liés à un plantage, nous vous demanderons d'en fournir un. ]</br>

### Version du micrologiciel du système

X.X.X</br>
[Remplacez les X par la version du micrologiciel du système au moment du plantage. ]</br>
[ Vous pouvez trouver la version de votre firmware dans Paramètres -> Système, sous "Mise à jour du système". ]</br>
[ S'il indique "Mise à jour en attente", vous pouvez effacer la mise à jour en attente en redémarrant en mode maintenance. ]</br>

### Environnement?

- Quel bootloader (fusèe, hekate, etc) a été lancé par Atmosphère :
- Version officielle ou version non officielle :
  - [ Version officielle x.x.x (ou) version non officielle ]
  - [ Si vous utilisez une version non officielle, indiquez où/comment vous avez acquis la version. ]
     - [ Ex : Auto-compilation ]
     - [ Ex : Distribution d'Atmosphère par Kosmos ]
- Avez-vous des kips ou des sysmodules supplémentaires que vous chargez :
- Logiciel Homebrew installé : [ * ]
- EmuMMC ou SysNAND :
  - [ Si vous utilisez un EmuMMC, indiquez s'il est basé sur une partition ou sur un fichier. ]

### Contexte supplémentaire ?

- Informations supplémentaires sur votre environnement :
- [ Toute autre information pertinente à votre problème. 
[image]
]

masagrator, 2022-05-01:
Did you just copy-paste what you previously wrote, but this time in French, put it into middle of issue template without editing it at all, put screenshot at the end and called it a day?

closed 2022-05-01 by SciresM

lsp199308, 2022-05-01:
This is not ams error, you have a corrupt prondinfo, please check
https://github.com/CaramelDunes/prodinfo_gen/issues/24