Atmosphere-NX/Atmosphere #1871: failed to stat stage2 (atmosphere/fusee-secondary.bin) after updating firmware to 14.1.0

Lectro72, 2022-05-02:
## Bug Report
So i was updating my firmware with daybreak, i selected "Install (FAT32 + exFAT)" as i was told that was the right option.
After receiving a message "Update applied succesfully", i went up and rebooted my switch, but got the following crash report:

"Fatal Error : [NXBOOT] Unable To Identify Package1"

I tried different solutions and none of them worked. I installed the latest atmosphere update, also downloaded the latest sigpatches, but nothing. After searching, I saw that apparently deleting my whole atmosphere file and replacing it with the latest version would do the trick, but now i have this new crash report:

"failed to stat stage2 (atmosphere/fusee-secondary.bin)"

I tried different solutions to similar problems in this thread but none of them work. Can anyone help me? Thank you very much in advance.

(I use the latest version of hekate, 5.7.2)

closed 2022-05-02 by SciresM

Lectro72, 2022-05-02:
> Update fusee
Could you provide me with a link? 
I can't seem to find the latest version of fusee, or rather I seem to not fully understand how it works

SciresM, 2022-05-02:
The issues are not for support, they are for bugs.

Fusee is downloaded from the same place as atmosphere release zips -- in the releases section of this github.