Atmosphere-NX/Atmosphere #1872: Keep getting boot errors after updating firmware and fusee.

Lectro72, 2022-05-02:
## Bug Report
I don't know where to put this as i haven't found a support section, so im sorry if this is a bother.

I updated my firmware to 14.1.1, and also updated my fusee. I apparently no longer use Fusee-primary.
Still, no matter what I try, i still get this message:

"failed to stat stage2 (atmosphere/fusee-secondary.bin)"

I updated things on atmosphere, on payloads, on root directory, on switch, and nothing works, i still keep getting that message.
If i try to inject a different payload, like "Fusee.bin" it just says:

"Incorrect fusee version! (program=0xb6b6ac70, mtc=0xb4dcf71)"

I really don't understand what am I doing wrong, i've been almost all the afternoon trying to solve this.

SciresM, 2022-05-02:
I already closed your previous issue and you are *still* not using the issue templates. You need to use the latest release and fusee together. Seek support on discord.

SciresM, 2022-05-02:
The issue tracker is not for support. I told you this already.

closed 2022-05-02 by SciresM