Atmosphere-NX/Atmosphere #1873: updated switch while running atmosphere on accident

negative-son, 2022-05-02:
## Bug Repot

every time i try to run atmosphere now i get  this a  error with this file report
[report_00000000094f6110.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8606309/report_00000000094f6110.zip) 
[image]

SciresM, 2022-05-02:
1) you did not use the issue templates
2) remove or update mission control

closed 2022-05-02 by SciresM

negative-son, 2022-05-02:
sorry but what is mission control

SciresM, 2022-05-02:
It is a custom system module you have installed. Remove it.

negative-son, 2022-05-03:
i deleted  mission control but i still get a error but there not gray screen

SciresM, 2022-05-03:
The issue tracker is for bugs, not support. Get further help on discord.