Atmosphere-NX/Atmosphere #1875: switch stopped working when opening gallery while running NRO with title takeover

Kennnwxyz, 2022-05-05:
## Bug Report

### What's the issue you encountered?

[My switch (14.1FW, Atmopshere 1.3.2) will stop working when opening gallery while running NRO in title takeover mode (R+Launch game)]</br>

### How can the issue be reproduced?

1. R+Start game 
2. Launch NRO, in my case it's retroarch and moonlight-switch
3. go back to home
4. open gallery
5. switch would just stop working, cant do anything but still on. atmosphere did not crash so no crash report. home and power button doesn't work

 hold power button for 10 second to hard reboot to fix

### Crash Report

[ N/A ]</br>

### System Firmware Version

14.1.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate v5.7.2
- Official release or unofficial build: official
- Do you have additional kips or sysmodules you're loading: I dont think so
- Homebrew software installed: [ MissionControl for third party controller ]
- EmuMMC or SysNAND: Partition-based emuMMC
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?
- Additional info about your environment: I was trying to open screenshot gallery while title takeover NRO but forgot to press R, and my switch just suddenly stop working and it can be replicated everytime

SciresM, 2022-05-06:
This sounds to me like the homebrew you're using is causing a service exhaustion of some kind, which prevents hbl from launching twice concurrently.

What custom sysmodules do you have? Please post a screenshot of your atmosphere/contents folder. This is likely not an Atmosphere issue, so we should diagnose appropriately.

Kennnwxyz, 2022-05-06:
Would this be the correct folder? I can't even remember what I put in here, might have to check 1 by 1
[image]

SciresM, 2022-05-06:
Yeah, that is a lot of custom sysmodules. 

There's nothing to be done with that many -- there'll be some instability. In this case I believe you're running out of some service's handles.

closed 2022-05-06 by SciresM